avafinger/nanopi-m4-ubuntu-base-minimal #1: after flash sd and reboot then can't see anything on screen with hdmi

oiramario, 2018-12-18:
at first, thanks for your work!
my nanopi M4 was working with firefly fireware.
i followed your steps, and flash is fine, but the screen(hdmi) is black after reboot. 
pls give your advice.
thanks!

avafinger, 2018-12-18:
Do you see the heartbeat on the board?
The video screen enters in sleep mode if nothing is typed on the keyboard.

Try this:

a) Hit ENTER twice right after boot

b) Use this DTB (rk3399-nanopi4-rev01.dtb attached)
The reason for this new **dtb** is that on v1.1 you have **a dual header monitor** and **HDMI** is the second monitor, even if eDP LCD is not attached. So the main screen is the LCD which is not attached.

[rk3399-nanopi4-rev01.dtb.zip](https://github.com/avafinger/nanopi-m4-ubuntu-base-minimal/files/2690646/rk3399-nanopi4-rev01.dtb.zip)

overwritte th on in your **/boot**

avafinger, 2018-12-18:
If you don't see the heartbeat I strongly suggest you try with another SD card.
It means the kernel was not loaded. Chances are you have a BAD sd card.

oiramario, 2018-12-18:
I flash firefly fireware back with the same SD card, it's worked.
I tried flash with SD card and emmc, cannot work both of them.
:( anything wrong with it?

oiramario, 2018-12-18:
could you please make an img with them?
so i can use win32diskimager with it.

avafinger, 2018-12-18:
ok. i will.

Just a question, do you try to boot from SD card with eMMC attached?

oiramario, 2018-12-18:
M4 boot first with SD card, second with eMMC.
I tried both of them without another attached.

avafinger, 2018-12-18:
hmm, looks weird.. you should boot with no eMMC attached. The boot sequence is for the SD card.
I will provide the IMG asap but this scenario does not look promising.

If you are an experienced user/developer you could attach a USB-debug (minicom 1500000 8N1 | NOR | Minicom 2.7 | VT102 | Offline) and grab the boot log.

BTW, don't know about yout **firefly fireware**

oiramario, 2018-12-18:
Oops... it's firmware.
just like this: rk3399-sd-friendlydesktop-bionic-4.4-arm64-YYYYMMDD.img.zip

avafinger, 2018-12-18:
Can you tell how did you flash sd card and eMMC?

oiramario, 2018-12-18:
[image]
[image]
[image]
Screenshots attached.

avafinger, 2018-12-18:
Just to be clear about the **snap** mention.
/boot should be **/dev/sdc1** and not **/dev/loop17**

While **img** is not ready and if you can try to flash it in another Linux box (without the snap &  /dev/loopxx) woul be great.

oiramario, 2018-12-19:
Yesterday I'm working on VMware workstation. And now it's worked when I flash it to SD card on another Linux box. Great!!!
However I tested to flash to eMMC that cannot work yet. I think IMG is necessary.

avafinger, 2018-12-19:
How did you try to flash to eMMC? I think there is a special partition to load the BSP kernel (BSP u-boot and BSP kernel from the manufacturer) and that needs to be wiped out, but if you do you will not be able to boot with the manufacturer img again without restoring this bootloader part and for that, you need Rockchip tool.

Here is the **img** you can try out: https://mega.nz/#!YLZhlYJb!uqvCIjpeBoGyfhaNYPJwb67kma7wmGTn5ZObe-CPrb4

Technically there is no difference in using the **,/flash_sd.sh** and **img**, with some advantages to the flash_sd.sh that writes only 800MB and **img** will write 800MB plus 7.2GB of zeros, that means the script is 3x faster and burn less. (in theory only).

You can try the **img** and see how it goes. If it does not work and you decide to wipe out the eMMC you should be sure you can see the eMMC partitions from a clean sd boot before you wipe out eMMC contents.

oiramario, 2018-12-19:
Boot Linux from SD card and flash it to eMMC attached on nanopi M4.
sudo ./flash_sd.sh /dev/mmcblk1 (on M4, SD card is mmcblk0, eMMC is mmcblk1).

Before try flash the img, one more thing...
Without the rk3399-nanopi4-rev01.dtb you attached above, screen shows double shift when run glesmark. My suggestion is update it within next release. 
[image]
[image]

avafinger, 2018-12-19:
I did this to Release v1.3 , there is Kodi already installed.

One more thing to note is /etc/fstab, rootfs points to /dev/mmcblk0p2 , maybe you flash again to eMMC as you did and before reboot you edit the file (in eMMC partition) e change it to /dev/mmcblk**1**p2 and then sync and rebbot. I think it will work!

oiramario, 2018-12-19:
Using the ,/flash_sd.sh and img for Kodi v1.3 and minimalI v1.1 I tested that is work only to SD card.
But I cannot see the eMMC attached when type df -h.
I will try the method you suggested.

One more thing...
Would you please provide the GBM/DRM to me? I need them to cross-compile for implement zero-copy in ogles.

oiramario, 2018-12-20:
Do you tested kmscube that is working?
The kmscube you released(614KB) cannot running.
I build my own kmscube(47KB) shows cannot load all the modules.

[image]

oiramario, 2018-12-20:
https://github.com/austriancoder/kmscube.git
This one i'm using.
Test on your kodi v1.3.

avafinger, 2018-12-20:
It is not implemented in Mali. Remove (comment) the code.

Can you post the **ls /dev/mmcblk*** result? so i can try to write the script for eMMC.

oiramario, 2018-12-20:
I'll post the results tomorrow asap.

avafinger, 2018-12-20:
To be able to run kmscube:

wget https://github.com/avafinger/nanopi-m4-ubuntu-base-minimal/releases/download/kmscube_gbm_v1/kmscube
sudo chmod +x kmscube 

then:
./kmscube

avafinger, 2018-12-21:
Okay, you could try this:

* mkdir -p sd
* copy /etc/fstab to this dir and edit the file fstab and change all  mmcblk**0** to mmcblk**1**
* use the script to flash to eMMC as you did before
* after flashing eMMC you mount it if not mounted:
```
mkdir -p emmc
sudo mount /dev/mmcblk1p2 emmc
```
copy your modified fstab to ./emmc/etc/ to overwrite the file ./emmc/etc/stab , make sure is the one you edited.
shutdown and boot without sd card.

You could do it with v1.1 which is lighter than v1.3..

avafinger, 2018-12-21:
All instructions above assume you booted from SD CARD with eMMC attached, and not using an emmc adapter. If you use emmc adapter just edit the /etc/fstab on the emmc and changed it.

oiramario, 2018-12-21:
- run linux on SD card
- flash to eMMC, sudo ./flash_sd.sh /dev/mmcblk1
- modified fstab@./emmc/etc, change all mmcblk0 to mmcblk1
- remove SD card and reboot from eMMC
- no heartbeat

[image]

avafinger, 2018-12-21:
Ok, unfortunately, bootloader looks for Android /boot style.
We need to clean eMMC to get rid of that 3d partition but if something goes wrong you end up needing the RK tools to recover.
Let me brush up the process and you could try again.

oiramario, 2018-12-21:
The firmware of friendlyarm could be download here: http://download.friendlyarm.com/NanoPiM4.
Write the img to SD card with win32diskimager. 
Insert this card into your board's boot slot and power on. Then it will flash to eMMC. After board has booted I checked etc/fstab in SD card however it's empty.
FYI.

oiramario, 2018-12-24:
kmscube print "undefined symbol:gbm_create_device" with nanopi-m4-ubuntu-base-minimal v1.1.

oiramario, 2018-12-24:
Here is an eMMC flasher images for you:
https://drive.google.com/drive/folders/1p02hbz2ongMLnv5fTraKQYtm3VwYVK2e/emmc-flasher-images.tgz

[image]

avafinger, 2018-12-24:
you need to setup **gbm**. the kmscube runs on **gbm**

Refer: https://github.com/avafinger/nanopi-m4-ubuntu-base-minimal/releases/tag/v1.2.1

```
sudo apt-get install libjpeg-turbo8 libjpeg8 libpng16-16 libegl1 libegl-mesa0 libpng-dev libjpeg-dev
sudo apt-get install libglu1-mesa libglapi-mesa libgles2-mesa-dev libglu1-mesa-dev libegl1-mesa-dev

```
**sudo apt-get install libgbm-dev**

cd /usr/rk3399-libs/lib64/
sudo chmod +x  mali-config.sh
sudo  ./mali-config.sh and set **gbm**
sudo shutdown -h now (wait and power cicle the borad)

Once you setup the **gbm** mali support you can only run **gbm** apps and not **fbdev only** anymore.

[image]

oiramario, 2018-12-24:
Thanks, I'll try.
URL above is [https://drive.google.com/drive/folders/1p02hbz2ongMLnv5fTraKQYtm3VwYVK2e](url), and you can download emmc-flasher-images.tgz.

oiramario, 2018-12-24:
Thanks, I'll try.
URL above is [,](https://drive.google.com/drive/folders/1p02hbz2ongMLnv5fTraKQYtm3VwYVK2e) and you can download emmc-flasher-images.tgz.

oiramario, 2018-12-24:
Thanks, I'll try.
URL above is [https://drive.google.com/drive/folders/1p02hbz2ongMLnv5fTraKQYtm3VwYVK2e](https://drive.google.com/drive/folders/1p02hbz2ongMLnv5fTraKQYtm3VwYVK2e) and you can download emmc-flasher-images.tgz.

oiramario, 2018-12-27:
Now i make everything fine.
This issue has been solved.
Thanks for your work.
:+1:

oiramario, 2019-01-01:
I wrote my own egl + gbm code test it's worked on your kodi v1.3.
While I can do that on mini v1.1 with some apt-get install / dpkg -i / ln -s.
I make a cross-compile tool chain in ubuntu 18.04 on x86_64, write code using vscode + cmake + gdbserver for rk3399.
Thanks for your work again.

avafinger, 2019-01-02:
Glad it worked for you. And how about the eMMC flashing?

avafinger, 2019-01-02:
**The eMMC booting issue has been fixed!**

OS Image and Kodi 18.0 RC5 will be in the v1.8 release.

avafinger, 2019-01-03:
Install to eMMC is available at https://github.com/avafinger/nanopi-m4-ubuntu-base-minimal/releases/tag/v1.8

closed 2019-01-03 by avafinger

oiramario, 2019-01-03:
Awesome! I have successfully burned v1.8 to eMMC.
Thanks for your work. :+1: 
Happy new year!  :-)