avafinger/nanopi-m4-ubuntu-base-minimal #2: Kernel modifications (TUN)

lorefx, 2019-01-06:
Hello, I really appreciate your effort, and I just flashed the v1.4 on my nanopi m4. Installs and run like a charm, but I have a question: could it be possible to modify/recompile the kernel so I can add the TUN module or other modules in the future? I'd really like to use it as an openvpn gateway in my house.
Thanks

avafinger, 2019-01-07:
Sure. Please provide which modules are needed for the openvpn gateway. I will add on next release of 4.4.170.

lorefx, 2019-01-07:
Well, I said other modules to let anyone add something to their own kernel without requiring you to make any work or further release and wothout pollluting everyone's kernel with something needed only by a user or two. At the moment I'll only need to work with tun because openvpn requires it to activate the encrypted connection.The other requirement is iptables in kernel to allow network traffic rules. 
Thanks

oiramario, 2019-01-09:
Agree, I have similar needs.
Provide a minimal clean system, only gpu and vpu hardware acceleration environment, we can flash to sd card or eMMC, and then we can easily add various modules.

avafinger, 2019-01-13:
Kernel 4.4.170 is out.

**sudo modprobe tun**
```
[  562.921825] tun: Universal TUN/TAP device driver, 1.6
[  562.921841] tun: (C) 1999-2004 Max Krasnyansky <maxk@qualcomm.com>

```

I have been playing with kernel 5.0-rc1 and seems very stable, maybe this kernel would fit better the openvpn and iptables desire. Unfortunately no Wifi, GPU and VPU available for this kernel yet.

I am uploading 4.4.170-rk3399 on v1.9, backup your data and upgrade.

oiramario, 2019-01-16:
@avafinger Can you share your dockfile?

avafinger, 2019-01-16:
Sorry, what is a dockfile?

oiramario, 2019-01-16:
Oops... it's dockerfile...

oiramario, 2019-01-18:
I'll try this, thanks.

oiramario, 2019-01-23:
@avafinger If I use gcc 8.2, the compiled binary file does not match libmali.so(closed source), which makes it impossible to link. How do you solve it?

lorefx, 2019-01-24:
Hi, just a quick confirmation that the last version runs openvpn smoothly (as best as the nanopi will allow regarding speed). 
Unfortunately I couldn't find a way to load the iptables module as it seems not compiled into the kernel.
I'll try to build a custom kernel as you depicted in the previous message

Thanks again for your work

avafinger, 2019-01-24:
@oiramario 
I don't think gcc version would be the issue. Probably more related to finding the shared file libmali.so to be linked. Please, print here the link error and what you are doing.

@lorefx 
Yes, there is no iptables module compiled, i am afraid of breaking the build if i try to enable it.
If you don't need mali and vpu i think the best kernel to go with is the 5.0. I think it runs 10% faster, measured by feeling.. :)

lorefx, 2019-01-24:
I feel that mali and vpu are important to me also to run properly kodi
I'll try to work first with the 4.4 171 stable kernel

avafinger, 2019-01-28:
@lorefx 
Sorry, I think i forgot to enable TUN on 4.4.171, and linux 4.4.172 stable has been released, i will enable TUN by default.
There is support for ov13850 + ov4689 on 4.4.172.

avafinger, 2019-01-28:
**linux kernel 4.4.172-rk3399**
https://github.com/avafinger/nanopi-m4-ubuntu-base-minimal/releases/tag/v1.11

lorefx, 2019-02-02:
Hi, just 3 questions:
1) Could it be possible to have/retrieve your kernel .config for wach release so I can start from there to set the modules I need?
2) Every time you release a new image, dkpg -i will complain that another image is already present and that it can't overwrite existing files or directories. I can probably appen the force overwrite flag for dpkg, but isn't it something you also experience?
3) Using your image from scratch, kodi starts without hassles, but in the options it lacks completely the zeroconf and web server. Do you know something about that? Maybe it depends on some missing kernel modules or other installed packages?
4) To update kodi, is it simple a matter of apt-get install kodi?

Thanks for your time and effort!

avafinger, 2019-02-04:
1. I  will provide it asap.
2. I was not aware of this issue, i will check later and do something about it. Thank you for reporting it.
3. I don't know much about zeroconf and web server on KODI. I will learn and rebuild KODI when time permits.
4. No. KODI must be rebuild.

avafinger, 2019-02-08:
@lorefx 

Kernel 4.4.173 is out.
Try https://github.com/avafinger/nanopi-m4-ubuntu-base-minimal/releases/tag/v1.13
Fixed the install issue.

lorefx, 2019-02-23:
Hi, sorry for being a little late but only today I managed to take out the nanopi and do things:)
First of all I installe the 4.4.175 and everything went well.
Second I tried to build Kodi and it didn't succeeded (crashed as soon as I launched the compiled binary, I need to investigate it bettere I think or maybe I missed some steps. I think I'll open an issue only for that whenever I'll manage to take out some more tests).

In the meanwhile I want to notify to you that probably the only thing missing from your kodi release (The webserver) is built in by installing MicroHttpd before compiling kodi

sudo apt-get install libmicrohttpd-dev

So point number 3) is probably soved with this (again, cannot test it as my built kodi crashes).
Point number 2) is also solved

Do you managed to provide the kernel config file in the latest releases and onward? (Point 1)
Number 4) I'll have to look better in your guide and xbmc guide I think

Thanks!!

avafinger, 2019-03-23:
https://github.com/avafinger/nanopi-m4-ubuntu-base-minimal/releases/tag/v1.19

Kernel 4.4.177-rk3399 has TUN enabled as module.

closed 2019-03-23 by avafinger

TCB13, 2019-06-29:
I'm stuck with 4.4.174 and there is no indication of a tun module available (also nano pi m4):

````
sudo modprobe tun
modprobe: ERROR: ../libkmod/libkmod.c:586 kmod_search_moddep() could not open moddep file '/lib/modules/4.4.174-rk3399/modules.dep.bin'
modprobe: FATAL: Module tun not found in directory /lib/modules/4.4.174-rk3399
````