avafinger/nanopi-m4-ubuntu-base-minimal #4: Re-create uboot images

dmikushin, 2019-02-09:
Hi @avafinger,

I've noticed you have released uboot and trust images along with v1.8 eMMC version. I know that booting from eMMC requires a different command line in uboot env. So I'm wondering if you possibly have helpful scripts for cooking uboot from scratch for RK3399, that can be shared?

Many thanks!

avafinger, 2019-02-10:
@dmikushin ,
I built the u-boot manually following the rockchip script commands and using their blobs, that's why i did not touch trust,img and boot,img for a long time (it is from oct 2018) . I had a hard time finding the correct version of bl31 that boots and runs the board stable. The disadvantage of using those blobs is that seems to limit the DDR speed. If you build bl31 from the sources you can use a newer u-boot version.
If you benchmark the memory speed you will see lower values than other distros but the board runs much more stable.

Regarding u-boot for eMMC or SD, no changes needed,  only the env changed to match the boot device order.

dmikushin, 2019-02-10:
@avafinger thanks for reply! OK, from `flash_kodi_emmc.sh` it's now clear that you use a different boot scheme with ext4 boot partition. This is quite interesting. On [M4 wiki](http://wiki.friendlyarm.com/wiki/index.php/NanoPi_M4) uboot loads kernel image directly using uboot environment to provide kernel arguments. I've been out of luck trying to alter uboot env arguments - they are always reset to defaults after first boot. As result, it's not possible to put uboot into interactive mode or get anything practical out of it. Maybe you have any pointers on this?

Sorry, I got confused: is it **your** uboot the one, which is clocking DDR lower for stability? Or the one from M4 wiki?

avafinger, 2019-02-11:
> Sorry, I got confused: is it **your** uboot the one, which is clocking DDR lower for stability? Or the one from M4 wiki?

Rockchip blob. From rockchip official support.  Use default u-boot, not the Android like u-boot.

avafinger, 2020-08-29:
@dmikushin 

If you are looking for the latest u-boot , here i have instructions for that : https://github.com/avafinger/nanopi-m4-ubuntu-base-minimal/blob/master/README.md#mainline-u-boot

I used the latest release: https://github.com/avafinger/nanopi-m4-ubuntu-base-minimal/blob/master/README.md#kodi-mainline-kernel-579

closed 2020-08-29 by avafinger