avafinger/nanopi-m4-ubuntu-base-minimal #6: Please release a version of ubuntu 18.04 including all recent updates

oiramario, 2019-02-15:
Linux kernel 4.4.174
u-boot rc1
without kodi
libmali + mpp + libdrm + gstreamer + ...
flash.sh with eMMC && SD card
anything else...

I am trying to write a Dockerfile script to automatically build the above content, not all done, if you can, I hope you can give advice, thank you.
[https://github.com/oiramario/nanopi-m4-ubuntu-base-dockfile](url)
thanks every much.

avafinger, 2019-02-15:
Ok.
You could have a look at Armbian, i think there is a build for Dockerfile, but i am not sure.
If I understand correctly you want a minimal Image without all this stuff (kodi + libmali + mpp + libdrm + gstreamer + ...), right?

oiramario, 2019-02-15:
Just don't need kodi, other hardware-accelerated libraries are required.

oiramario, 2019-02-18:
I wrote a program that needs to link libmali, which compiles normally under gcc7.
When I use gcc8 to compile it, I get the following error.

`/home/build/gcc-arm-8.2-2019.01-x86_64-aarch64-linux-gnu/bin/../lib/gcc/aarch64-linux-gnu/8.2.1/../../../../aarch64-linux-gnu/lib/libMali.so: undefined reference to `std::string::erase(unsigned long, unsigned long)@GLIBCXX_3.4'
/home/build/gcc-arm-8.2-2019.01-x86_64-aarch64-linux-gnu/bin/../lib/gcc/aarch64-linux-gnu/8.2.1/../../../../aarch64-linux-gnu/lib/libMali.so: undefined reference to `std::error_category::error_category()@GLIBCXX_3.4.15'
/home/build/gcc-arm-8.2-2019.01-x86_64-aarch64-linux-gnu/bin/../lib/gcc/aarch64-linux-gnu/8.2.1/../../../../aarch64-linux-gnu/lib/libMali.so: undefined reference to `vtable for std::basic_ostringstream<char, std::char_traits<char>, std::allocator<char> >@GLIBCXX_3.4'
/home/build/gcc-arm-8.2-2019.01-x86_64-aarch64-linux-gnu/bin/../lib/gcc/aarch64-linux-gnu/8.2.1/../../../../aarch64-linux-gnu/lib/libMali.so: undefined reference to `std::basic_ostream<char, std::char_traits<char> >& std::__ostream_insert<char, std::char_traits<char> >(std::basic_ostream<char, std::char_traits<char> >&, char const*, long)@GLIBCXX_3.4.9'
/home/build/gcc-arm-8.2-2019.01-x86_64-aarch64-linux-gnu/bin/../lib/gcc/aarch64-linux-gnu/8.2.1/../../../../aarch64-linux-gnu/lib/libMali.so: undefined reference to `VTT for std::basic_stringstream<char, std::char_traits<char>, std::allocator<char> >@GLIBCXX_3.4'
/home/build/gcc-arm-8.2-2019.01-x86_64-aarch64-linux-gnu/bin/../lib/gcc/aarch64-linux-gnu/8.2.1/../../../../aarch64-linux-gnu/lib/libMali.so: undefined reference to `__cxa_guard_acquire@CXXABI_1.3'
/home/build/gcc-arm-8.2-2019.01-x86_64-aarch64-linux-gnu/bin/../lib/gcc/aarch64-linux-gnu/8.2.1/../../../../aarch64-linux-gnu/lib/libMali.so: undefined reference to `std::__detail::_List_node_base::swap(std::__detail::_List_node_base&, std::__detail::_List_node_base&)@GLIBCXX_3.4.15'
`

Through google search, I learned that this is an ABI-related problem, so follow the prompts: **-D_GLIBCXX_USE_CXX11_ABI=0**, but the compiler still reports an error. 
How do you solve this problem?

oiramario, 2019-02-20:
@avafinger 
Which version of u-boot are you using? Where is the address of git? The u-boot I am currently using is 2019.1 downloaded from https://github.com/u-boot/u-boot. I don't find rk3399_defconfig in the configs directory. How do you compile it? Thank you!

oiramario, 2019-02-22:
I am going back to using gcc7 for a while, and I have already compiled the kernel and u-boot, thank you!

avafinger, 2019-08-03:
https://github.com/avafinger/nanopi-m4-ubuntu-base-minimal/releases/tag/1.24

closed 2019-08-03 by avafinger