avafinger/nanopi-m4-ubuntu-base-minimal #10: Kodi Rebuild procedure

lorefx, 2019-03-11:
Hello, I tried to follow the quick guide o how to rebuild kodi, but each time I try following the guide I always get stuck with an invalid executable causing a stack trace when executed.
For sure I'm missing some steps.

Would it be possible for you to update the readme with a detailed step-by-step on how to build kodi starting from a scratch flashed image of base nanopi m4 ubuntu base minimal?

Thanks!

avafinger, 2019-03-11:
> get stuck with an invalid executable causing a stack trace 

I used the guide and instructions from KODI team, never had this type of error. There was indeed a DRM issue but was fixed on latest kernel (should be).

Have you checked dmesg for possible errors?
Did you build ffmpeg or used the one i have built?

Please output the complete dmesg and Kodi log error (look inside ~/.kodi/temp).

I have compiled several times Kodi on SD CARD and i lost the sd card (wear out),  It's going to take some time for me to get back to compile kodi again.

Building Kodi is boring and tedious maybe there should be a way to cross-compile it in a host-pc, but i have not try it.

lorefx, 2019-03-12:
Ok, today I'll make a new try from scratch and document everything, so maybe it will be useful also on the official readme doc if we manage to sove it

lorefx, 2019-03-12:
Hi, I managed to put down the entire process starting from scratch (to reproduce it) and from the log it seems indeed a drm error.

---------------------------------------
PROCEDURE:

- Start from Release v1.3 to install on SD card
--- mkdir nanopi-m4
--- cd nanopi-m4
--- wget $(curl -s https://api.github.com/repos/avafinger/nanopi-m4-ubuntu-base-minimal/releases | grep -oP '"browser_download_url": "\K(.*)(?=")' | grep v1.3)
--- chmod 775 flash_kodi_sd.sh
--- sudo ./flash_kodi_sd.sh /dev/sdX (or /dev/mmcblkY) where X is a letter from b,c.. and Y is a number from 0,1..)

- Go to Release v1.18 step to update to latest
--- wget $(curl -s https://api.github.com/repos/avafinger/nanopi-m4-ubuntu-base-minimal/releases | grep -oP '"browser_download_url": "\K(.*)(?=")' | grep v1.18)
--- sudo dpkg -i linux-image-4.4.175-rk3399_1.0-8.deb (There is a typo)
--- sync && sudo reboot

- Update and upgrade system
--- sudo apt update
--- sudo apt upgrade
--- sync && sudo reboot

- At this point I made an SD card image to avoid repeating the setup steps

- Build Kodi: 
--- based on https://github.com/xbmc/xbmc/blob/master/docs/README.Linux.md#4-build-kodi
--- sudo apt-get install libmicrohttpd-dev (this is for web server in kodi)
--- git clone https://github.com/xbmc/xbmc kodi
--- cd kodi
--- sudo apt install autoconf automake autopoint gettext autotools-dev cmake curl default-jre gawk gcc g++ cpp gdc gperf libasound2-dev libass-dev libavahi-client-dev libavahi-common-dev libbluetooth-dev libbluray-dev libbz2-dev libcdio-dev libcec-dev libp8-platform-dev libcrossguid-dev libcurl4-openssl-dev libcwiid-dev libdbus-1-dev libegl1-mesa-dev libenca-dev libflac-dev libfontconfig1-dev libfmt-dev libfreetype6-dev libfribidi-dev libfstrcmp-dev libgcrypt20-dev libgif-dev libgles2-mesa-dev libglew-dev libglu1-mesa-dev libgnutls28-dev libgpg-error-dev libiso9660-dev libjpeg-dev liblcms2-dev liblirc-dev libltdl-dev liblzo2-dev libmicrohttpd-dev libmysqlclient-dev libnfs-dev libogg-dev libomxil-bellagio-dev libpcre3-dev libplist-dev libpng-dev libpulse-dev libsmbclient-dev libsqlite3-dev libssl-dev libtag1-dev libtiff5-dev libtinyxml-dev libtool libudev-dev libva-dev libvdpau-dev libvorbis-dev libxkbcommon-dev libxmu-dev libxrandr-dev libxslt1-dev libxt-dev netcat wayland-protocols lsb-release nasm python-dev python-pil python-minimal rapidjson-dev swig unzip uuid-dev yasm zip zlib1g-dev
--- sudo make -C tools/depends/target/crossguid PREFIX=/usr/local
--- sudo make -C tools/depends/target/flatbuffers PREFIX=/usr/local
--- sudo make -C tools/depends/target/libfmt PREFIX=/usr/local
--- sudo make -C tools/depends/target/wayland-protocols PREFIX=/usr/local
--- sudo make -C tools/depends/target/waylandpp PREFIX=/usr/local
--- cd ..
--- mkdir kodi-build
--- cd kodi-build
--- cmake ../kodi -DCMAKE_INSTALL_PREFIX=/usr/local -DCORE_PLATFORM_NAME=gbm -DGBM_RENDER_SYSTEM=gles
(Note: I used the gbm option as it was the most similar to the one in you readme)

//**************************************************//
./confgure log

-- The CXX compiler identification is GNU 8.2.0
-- The C compiler identification is GNU 8.2.0
-- The ASM compiler identification is GNU
-- Found assembler: /usr/bin/cc
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Source directory: /home/ubuntu/kodi
-- Build directory: /home/ubuntu/kodi-build
-- Generator: Single-configuration: Release (Unix Makefiles)
-- CMake Version: 3.10.2
-- System type: Linux
-- Checking to see if CXX compiler accepts flag -std=c++11
-- Checking to see if CXX compiler accepts flag -std=c++11 - yes
-- Found CXX11: -std=c++11  
-- Linker: GNU gold
-- Core system type: linux
[... 618 lines cut ...]
2019-03-12 14:35:01.654 T:547465715856  NOTICE:         m_deviceType      : AE_DEVTYPE_PCM
2019-03-12 14:35:01.654 T:547465715856  NOTICE:         m_channels        : FL, FR
2019-03-12 14:35:01.654 T:547465715856  NOTICE:         m_sampleRates     : 5512,8000,11025,16000,22050,32000,44100,48000,64000,88200,96000,176400,192000,384000
2019-03-12 14:35:01.655 T:547465715856  NOTICE:         m_dataFormats     : AE_FMT_U8,AE_FMT_S16NE,AE_FMT_S24NE3,AE_FMT_S24NE4,AE_FMT_S32NE,AE_FMT_FLOAT
2019-03-12 14:35:01.655 T:547465715856  NOTICE:         m_streamTypes     : No passthrough capabilities
2019-03-12 14:35:01.655 T:547465715856  NOTICE:     Device 3
2019-03-12 14:35:01.655 T:547465715856  NOTICE:         m_deviceName      : alsa_output.platform-rt5651-sound.stereo-fallback
2019-03-12 14:35:01.655 T:547465715856  NOTICE:         m_displayName     : Built-in Audio Stereo
2019-03-12 14:35:01.655 T:547465715856  NOTICE:         m_displayNameExtra: Analog Output (PULSEAUDIO)
2019-03-12 14:35:01.655 T:547465715856  NOTICE:         m_deviceType      : AE_DEVTYPE_PCM
2019-03-12 14:35:01.655 T:547465715856  NOTICE:         m_channels        : FL, FR
2019-03-12 14:35:01.655 T:547465715856  NOTICE:         m_sampleRates     : 5512,8000,11025,16000,22050,32000,44100,48000,64000,88200,96000,176400,192000,384000
[... cut ...]
2019-03-12 14:35:01.656 T:547465715856  NOTICE:         m_sampleRates     : 5512,8000,11025,16000,22050,32000,44100,48000,64000,88200,96000,176400,192000,384000
2019-03-12 14:35:01.656 T:547465715856  NOTICE:         m_dataFormats     : AE_FMT_U8,AE_FMT_S16NE,AE_FMT_S24NE3,AE_FMT_S24NE4,AE_FMT_S32NE,AE_FMT_FLOAT
2019-03-12 14:35:01.656 T:547465715856  NOTICE:         m_streamTypes     : No passthrough capabilities
2019-03-12 14:35:01.667 T:547457323152  NOTICE: PulseAudio: Opened device Default in pcm mode with Buffersize 199 ms
2019-03-12 14:35:01.702 T:547552133136   ERROR: CWinSystemGbm::InitWindowSystem - failed to initialize Legacy DRM

avafinger, 2019-03-12:
@lorefx 
Your procedure is correct, the error is the DRM issue. For some weird reason, the drm patch did not get into 4.4.175-rk3399.
Please, keep this build and doc and as soon as i get 4.4.177 release you upgrade and run Kodi.

lorefx, 2019-03-15:
I was thinking that the strange thing is that the kodi version shipped by you continues to start normally, and it shouldn't if it were a kernel issue. Don't you agree? Am I missing something?

avafinger, 2019-03-23:
@lorefx 
There is a workaround in Kernel 4.4.177-rk3399.
Please try this kernel and see if your kodi build works:
https://github.com/avafinger/nanopi-m4-ubuntu-base-minimal/releases/tag/v1.19

closed 2019-03-23 by avafinger

lorefx, 2019-03-24:
I just remade the entire procedure from scratch, and something improved on the DRM but still not working. At the moment I cannot test anymore but here is my output. Ill try sine search for the issue tomorrow.

ubuntu@nanopi-m4:~/kodi-build$ ./kodi-gbm
libva info: VA-API version 1.1.0
libva info: va_getDriverName() returns -1
libva error: va_getDriverName() failed with unknown libva error,driver_name=(null)
Bus error

This is the kodi log
ubuntu@nanopi-m4:~/kodi-build$ cat /home/ubuntu/.kodi/temp/kodi.log
2019-03-24 18:35:36.393 T:547752501264  NOTICE: -----------------------------------------------------------------------
2019-03-24 18:35:36.393 T:547752501264  NOTICE: Starting Kodi (18.2-RC1 Git:20190323-98d18d0a1e). Platform: Linux ARM 64-bit
2019-03-24 18:35:36.393 T:547752501264  NOTICE: Using Release Kodi x64 build
2019-03-24 18:35:36.393 T:547752501264  NOTICE: Kodi compiled 2019-03-24 by GCC 8.2.0 for Linux ARM 64-bit version 4.15.18 (266002)
2019-03-24 18:35:36.393 T:547752501264  NOTICE: Running on Ubuntu 18.04.2 LTS, kernel: Linux ARM 64-bit version 4.4.177-rk3399
2019-03-24 18:35:36.396 T:547752501264  NOTICE: FFmpeg version/source: c8b57d4333
2019-03-24 18:35:36.396 T:547752501264  NOTICE: 6 CPU cores available
2019-03-24 18:35:36.396 T:547752501264  NOTICE: ARM Features: Neon enabled
2019-03-24 18:35:36.396 T:547752501264  NOTICE: special://xbmc/ is mapped to: /usr/local/share/kodi
2019-03-24 18:35:36.396 T:547752501264  NOTICE: special://xbmcbin/ is mapped to: /home/ubuntu/kodi-build
2019-03-24 18:35:36.396 T:547752501264  NOTICE: special://xbmcbinaddons/ is mapped to: /home/ubuntu/kodi-build/addons
2019-03-24 18:35:36.396 T:547752501264  NOTICE: special://masterprofile/ is mapped to: /home/ubuntu/.kodi/userdata
2019-03-24 18:35:36.396 T:547752501264  NOTICE: special://envhome/ is mapped to: /home/ubuntu
2019-03-24 18:35:36.396 T:547752501264  NOTICE: special://home/ is mapped to: /home/ubuntu/.kodi
2019-03-24 18:35:36.396 T:547752501264  NOTICE: special://temp/ is mapped to: /home/ubuntu/.kodi/temp
2019-03-24 18:35:36.396 T:547752501264  NOTICE: special://logpath/ is mapped to: /home/ubuntu/.kodi/temp
2019-03-24 18:35:36.396 T:547752501264  NOTICE: The executable running is: /home/ubuntu/kodi-build/kodi-gbm
2019-03-24 18:35:36.396 T:547752501264  NOTICE: Local hostname: nanopi-m4
2019-03-24 18:35:36.396 T:547752501264  NOTICE: Log File is located: /home/ubuntu/.kodi/temp/kodi.log
2019-03-24 18:35:36.396 T:547752501264  NOTICE: -----------------------------------------------------------------------
2019-03-24 18:35:36.402 T:547752501264    INFO: loading settings
2019-03-24 18:35:36.404 T:547752501264  NOTICE: special://profile/ is mapped to: special://masterprofile/
2019-03-24 18:35:36.429 T:547752501264   DEBUG: CSkinSettings: no <skinsettings> tag found
2019-03-24 18:35:36.436 T:547752501264  NOTICE: No settings file to load (special://xbmc/system/advancedsettings.xml)
2019-03-24 18:35:36.436 T:547752501264  NOTICE: No settings file to load (special://masterprofile/advancedsettings.xml)
2019-03-24 18:35:36.437 T:547752501264  NOTICE: Default Video Player: VideoPlayer
2019-03-24 18:35:36.437 T:547752501264  NOTICE: Default Audio Player: paplayer
2019-03-24 18:35:36.437 T:547752501264  NOTICE: Disabled debug logging due to GUI setting. Level 0.
2019-03-24 18:35:36.437 T:547752501264  NOTICE: Log level changed to "LOG_LEVEL_NORMAL"
2019-03-24 18:35:37.985 T:547752501264  NOTICE: PulseAudio: Server found running - will try to use Pulse
2019-03-24 18:35:38.016 T:547752501264  NOTICE: Running database version Addons27
2019-03-24 18:35:38.617 T:547752501264  NOTICE: ADDON: audiodecoder.2sf v2.0.0 installed
2019-03-24 18:35:38.617 T:547752501264  NOTICE: ADDON: audiodecoder.asap v2.0.0 installed
2019-03-24 18:35:38.617 T:547752501264  NOTICE: ADDON: audiodecoder.dumb v2.0.0 installed
2019-03-24 18:35:38.617 T:547752501264  NOTICE: ADDON: audiodecoder.fluidsynth v2.0.0 installed
2019-03-24 18:35:38.617 T:547752501264  NOTICE: ADDON: audiodecoder.gme v2.0.0 installed
2019-03-24 18:35:38.617 T:547752501264  NOTICE: ADDON: audiodecoder.gsf v2.0.0 installed
2019-03-24 18:35:38.617 T:547752501264  NOTICE: ADDON: audiodecoder.modplug v2.0.0 installed
2019-03-24 18:35:38.618 T:547752501264  NOTICE: ADDON: audiodecoder.ncsf v2.0.0 installed
2019-03-24 18:35:38.618 T:547752501264  NOTICE: ADDON: audiodecoder.nosefart v2.0.0 installed
2019-03-24 18:35:38.618 T:547752501264  NOTICE: ADDON: audiodecoder.openmpt v2.0.1 installed
2019-03-24 18:35:38.618 T:547752501264  NOTICE: ADDON: audiodecoder.organya v1.1.3 installed
2019-03-24 18:35:38.618 T:547752501264  NOTICE: ADDON: audiodecoder.qsf v2.0.0 installed
2019-03-24 18:35:38.618 T:547752501264  NOTICE: ADDON: audiodecoder.sidplay v1.1.3 installed
2019-03-24 18:35:38.618 T:547752501264  NOTICE: ADDON: audiodecoder.snesapu v2.0.0 installed
2019-03-24 18:35:38.618 T:547752501264  NOTICE: ADDON: audiodecoder.ssf v2.0.0 installed
2019-03-24 18:35:38.618 T:547752501264  NOTICE: ADDON: audiodecoder.timidity v2.0.1 installed
2019-03-24 18:35:38.618 T:547752501264  NOTICE: ADDON: audiodecoder.upse v2.0.0 installed
2019-03-24 18:35:38.618 T:547752501264  NOTICE: ADDON: audiodecoder.usf v2.0.0 installed
2019-03-24 18:35:38.618 T:547752501264  NOTICE: ADDON: audiodecoder.vgmstream v1.1.3 installed
2019-03-24 18:35:38.618 T:547752501264  NOTICE: ADDON: audiodecoder.wsr v2.0.0 installed
[... cut ...] reate_context EGL_KHR_surfaceless_context EGL_KHR_gl_texture_cubemap_image EGL_EXT_create_context_robustness EGL_KHR_cl_event2
2019-03-24 18:35:40.198 T:547752501264  NOTICE: EGL_CLIENT_EXTENSIONS = EGL_EXT_client_extensions EGL_EXT_platform_base EGL_KHR_client_get_all_proc_addresses EGL_KHR_platform_gbm EGL_MESA_platform_gbm
2019-03-24 18:35:40.231 T:547752501264  NOTICE: Checking resolution 16
2019-03-24 18:35:40.231 T:547752501264 WARNING: CGBMUtils::DestroySurface - surface already destroyed

avafinger, 2019-03-25:
Does it crash afterthe last part?
```
2019-03-24 18:35:40.231 T:547752501264 NOTICE: Checking resolution 16
2019-03-24 18:35:40.231 T:547752501264 WARNING: CGBMUtils::DestroySurface - surface already destroyed
```

**It should have continued with:**
17:00:11.155 T:548151148560 WARNING: CGBMUtils::DestroySurface - surface already destroyed
```
17:00:11.192 T:548151148560  NOTICE: GL_VENDOR = ARM
17:00:11.192 T:548151148560  NOTICE: GL_RENDERER = Mali-T860
17:00:11.192 T:548151148560  NOTICE: GL_VERSION = OpenGL ES 3.2 v1.r14p0-01rel0-git(966ed26).2e3fa7564ebca70897f04bd9fb7bc67e
17:00:11.193 T:548151148560  NOTICE: GL_SHADING_LANGUAGE_VERSION = OpenGL ES GLSL ES 3.20
17:00:11.193 T:548151148560  NOTICE: GL_EXTENSIONS = GL_ARM_rgba8 GL_ARM_mali_shader_binary GL_OES_depth24 GL_OES_depth_texture GL_OES_depth_texture_cube_map GL_OES_packed_depth_stencil GL_OES_rgb8_rgba8 GL_EXT_read_format_bgra GL_OES_compressed_paletted_texture GL_OES_compressed_ETC1_RGB8_texture GL_OES_standard_derivatives GL_OES_EGL_image GL_OES_EGL_image_external GL_OES_EGL_image_external_essl3 GL_OES_EGL_sync GL_OES_texture_npot GL_OES_vertex_half_float GL_OES_required_internalformat GL_OES_vertex_array_object GL_OES_mapbuffer GL_EXT_texture_format_BGRA8888 GL_EXT_texture_rg GL_EXT_texture_type_2_10_10_10_REV GL_OES_fbo_render_mipmap GL_OES_element_index_uint GL_EXT_shadow_samplers GL_OES_texture_compression_astc GL_KHR_texture_compression_astc_ldr GL_KHR_texture_compression_astc_hdr GL_KHR_texture_compression_astc_sliced_3d GL_KHR_debug GL_EXT_occlusion_query_boolean GL_EXT_disjoint_timer_query GL_EXT_blend_minmax GL_EXT_discard_framebuffer GL_OES_get_program_binary GL_OES_texture_3D GL_EXT_texture_storage GL_EXT_multisampled_render_to_texture GL_OES_surfaceless_context GL_OES_texture_stencil8 GL_EXT_shader_pixel_local_storage GL_ARM_shader_framebuffer_fetch GL_ARM_shader_framebuffer_fetch_depth_stencil GL_ARM_mali_program_binary GL_EXT_sRGB GL_EXT_sRGB_write_control GL_EXT_texture_sRGB_decode GL_KHR_blend_equation_advanced GL_KHR_blend_equation_advanced_coherent GL_OES_texture_storage_multisample_2d_array GL_OES_shader_image_atomic GL_EXT_robustness GL_EXT_draw_buffers_indexed GL_OES_draw_buffers_indexed GL_EXT_texture_border_clamp GL_OES_texture_border_clamp GL_EXT_texture_cube_map_array GL_OES_texture_cube_map_array GL_OES_sample_variables GL_OES_sample_shading GL_OES_shader_multisample_interpolation GL_EXT_shader_io_blocks GL_OES_shader_io_blocks GL_EXT_tessellation_shader GL_OES_tessellation_shader GL_EXT_primitive_bounding_box GL_OES_primitive_bounding_box GL_EXT_geometry_shader GL_OES_geometry_shader GL_ANDROID_extension_pack_es31a GL_EXT_gpu_shader5 GL_OES_gpu_shader5 GL_EXT_texture_buffer GL_OES_texture_buffer GL_EXT_copy_image GL_OES_copy_image GL_EXT_color_buffer_half_float GL_EXT_color_buffer_float GL_EXT_YUV_target GL_OVR_multiview GL_OVR_multiview2 GL_OVR_multiview_multisampled_render_to_texture GL_KHR_robustness GL_KHR_robust_buffer_access_behavior GL_EXT_draw_elements_base_vertex GL_OES_draw_elements_base_vertex

```

lorefx, 2019-03-26:
Yes it does crash after the last part with the message in the console. Dmesg didn't show anything relevant

avafinger, 2019-03-26:
Things you should check:
* Make sure you don't have permission issues, check  "video"  access to "ubuntu" user. check: **id ubuntu** (try: **sudo kodi** amd not kodi-gbm )

* Check if GBM is working, use the **kmscube** sample.

* Enable LIBVA output: export LIBVA_TRACE=1

avafinger, 2019-03-26:
And build with:

> cmake ../kodi -DCMAKE_INSTALL_PREFIX=/usr/local -DCORE_PLATFORM_NAME=gbm -DGBM_RENDER_SYSTEM=gles -Dwith-ffmpeg=shared -Ddisable-x11=true