avafinger/nanopi-m4-ubuntu-base-minimal #11: Kernel Panic (probably) when trying to recompile kernel 4.4

lorefx, 2019-03-15:
I was trying to recompile the kernel to add my own modules. I followed your readme steps for kernel 5 and used the defconfig you passed to me.
What I obtain is (I think( a kernel panic on boot but I cannot confirm as I cannot see anythinh on the console so probably it hangs immediately.

These are the steps I did. Maybe I missed something? There are more steps you do when producing your releases? Could you write the steps you do?

Thanks!

//*************************************//
The steps are adapted for my will to recompile for 4.4

-    login into NanoPi M4

-    create directory

    mkdir -p linux
    cd linux  

 -   grab the latest kernel: https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/refs/tags

    wget https://cdn.kernel.org/pub/linux/kernel/v4.x/linux-4.4.176.tar.xz
    tar -xvpzf linux-4.4.176.tar.xz
    cd linux-linux-4.4.176

-    Build
    wget https://github.com/avafinger/nanopi-m4-ubuntu-base-minimal/files/2953801/nanopim4_defconfig.tar.gz
    tar -xvpzf nanopim4_defconfig.tar.gz
    cp nanopim4_defconfig arch/arm64/config
    make  defconfig
    make  menuconfig (added TUN)
    make  oldconfig
    make -j6 INSTALL_MOD_PATH=output Image 
    make -j6 INSTALL_MOD_PATH=output dtbs
    make -j6 INSTALL_MOD_PATH=output modules
    make -j6 INSTALL_MOD_PATH=output modules_install

-    Install

    export KV=$(strings ./arch/arm64/boot/Image |grep "Linux version"|awk '{print $3}')
    sudo cp -fv ./arch/arm64/boot/Image /boot/Image_${KV}
    sync
    sudo cp -vfr ./output/* /
    sync

 -   Make it current kernel

    cd /boot
    sudo ln -sf Image_${KV} Image
    cd ${KVD}
    sync

-    Reboot

    sudo shutdown -h now (remove power for 30 secs)
    **Boot** with your new kernel

- HANGS!!! :(

avafinger, 2019-03-15:
These instructions are for kernel 5.0.y only.
You can try rockchip kernel 4.4 (from rockchip github). Use menuconfig to set some ROCKCHIP specific modules.

Mainline kernel 4.4.177 is to be released soon, probably there will be a lot of fixes in this version.

avafinger, 2019-03-23:
@lorefx 
Mainline kernel 4.4.177 has been released.

Please, try your Kodi build with this kernel (make sure you have enough available space)
https://github.com/avafinger/nanopi-m4-ubuntu-base-minimal/releases/tag/v1.19

closed 2019-03-23 by avafinger