avafinger/nanopi-m4-ubuntu-base-minimal #14: Kodi Add-ons installed using apt are not recognized

myrkle11, 2019-11-14:
I installed the 1.24 release (Kodi 19 ALPHA1 NanoPi M4 RK3399), and while Kodi works without issue for the built-in add-ons, I can't get the apt installed add-ons to work -- Kodi doesn't recognize they've been installed. All the pvr add-ons suffer from this problem (pvr.hts for example).
I assume this is because of the custom build of Kodi, but in reading the Kodi build instructions, they don't talk about this issue. Do I need to build the add-on myself, or do you know of any way (changing configs, etc.) to have Kodi recognize the apt-installed add-ons?

avafinger, 2019-11-15:
I think the add-ons are version dependent. Try to find the add-on (zip file) for Kodi 19.x and see if you can install it.

myrkle11, 2019-11-15:
The version dependency makes sense, thanks. Unfortunately, I can't find an arm build of the add-on, just x86, etc. I will have to try building it, or trying a different image. Thanks for your help.