avafinger/nanopi-m4-ubuntu-base-minimal #15: Nanopi M4 emmc not booting

yousuffarhan, 2020-01-02:
I have followed the instructions, and my SBC is not booting. The green light turns on for a few seconds before shutting off. My monitor displays a black screen instead of the "no signal" messag,e

avafinger, 2020-01-02:
Do you have a TTL-USB to grab the boot log from the serial console?
Does it boot from SD Card?

yousuffarhan, 2020-01-03:
UPDATE: It boots but the display does not work. Anyway to install it headless? It's not letting me ssh.

avafinger, 2020-01-03:
The Image is a headless install, is that what you mean? There is no Desktop.
It sounds very weird. Do you have **NanoPi M4 v2**? There is no support for LPDDR4 in this Image.

No ssh means you have no eth0, the kernel has crashed, very likely.

If you don't have M4 v2, can you try this and see if it boots? https://github.com/avafinger/nanopi-m4-ubuntu-base-minimal/releases/tag/v1.27

avafinger, 2020-01-05:
You can try this kernel, no lpDDR4 yet.

Boot from older kernel (EON Image) and then apply:

* linux-image 5.4.1 Deb package      

      https://github.com/avafinger/nanopi-m4-ubuntu-base-minimal/releases/tag/v1.32

yousuffarhan, 2020-01-06:
Hey, so after a day of me leaving it to run, it has booted and has been stuck as "Starting Kernel Modules." 
It is the Nano Pi m4 V1, and ssh is not working.

avafinger, 2020-01-06:
Maybe it is related somehow to this: https://github.com/avafinger/nanopi-m4-ubuntu-base-minimal/issues/16#issuecomment-570996405