avafinger/nanopi-m4-ubuntu-base-minimal #16: Debian pkg kernel postinstall script fails due to wrong modules location

dmikushin, 2020-01-05:
Hi, got the following with the recent `linux-image-5.4.1_1.0-28.deb`:

```
ERROR: Could not update kernel! Try to fix it manually before reboot.
dpkg: error processing package linux-image-5.4.1 (--install):
 installed linux-image-5.4.1 package post-installation script subprocess returned error exit status 1
Errors were encountered while processing:
 linux-image-5.4.1
```

Noticed that the postinstall script checks for `/lib/modules/`, however, the package installs them to `/usr/lib/modules`.

avafinger, 2020-01-05:
This deb package is for Ubuntu **19.10** only. (Eoan)

If you are on Ubuntu 18.04 or older the deb package will not work for the reason you stated above.
But you can install the modules in /lib/modules/ manually, that will work for older Ubuntu or Debian.

dmikushin, 2020-01-05:
[EOAN](https://github.com/avafinger/nanopi-m4-ubuntu-base-minimal/releases/tag/v1.27) unfortunately does not boot. Tried to flash SD card twice - no boot, red LED only.

avafinger, 2020-01-05:
The installation was not finished, can you check the content of /boot? Should be something like this:

```
-rw-rw-r--  1 alex alex     1197 nov 14 14:22 boot.cmd
-rw-r--r--  1 root root     1269 nov 14 14:22 boot.scr
lrwxrwxrwx  1 root root       26 dez 17 21:27 dtb -> rk3399-nanopi-m4.dtb_5.4.4
lrwxrwxrwx  1 root root       11 dez 17 21:27 Image -> Image_5.4.4
-rw-r--r--  1 root root 25725440 nov 25 12:51 Image_5.4.0
-rw-r--r--  1 root root 25725440 nov 30 10:22 Image_5.4.1
-rw-r--r--  1 root root 25725440 dez  1 20:01 Image_5.4.1-rk3399
-rw-r--r--  1 root root 25725440 dez 17 21:26 Image_5.4.4
-rw-rw-r--  1 alex alex  1157052 nov 14 14:22 initrd.img
drwx------  2 root root    16384 nov 14 14:22 lost+found
-rw-r--r--  1 root root    55973 nov 25 12:54 rk3399-nanopi-m4.dtb_5.4.0
-rw-r--r--  1 root root    55973 nov 30 10:32 rk3399-nanopi-m4.dtb_5.4.1
-rw-r--r--  1 root root    55973 dez  1 20:02 rk3399-nanopi-m4.dtb_5.4.1-rk3399
-rw-r--r--  1 root root    55973 dez 17 21:27 rk3399-nanopi-m4.dtb_5.4.4

```

And the Kernel modules:

```
/usr/lib/modules
total 24
drwxr-xr-x  6 root root 4096 dez 17 21:26 .
drwxr-xr-x 76 root root 4096 dez  8 11:56 ..
drwxr-xr-x  3 root root 4096 nov 25 12:52 5.4.0
drwxr-xr-x  3 root root 4096 nov 30 09:21 5.4.1
drwxr-xr-x  3 root root 4096 dez  1 20:02 5.4.1-rk3399
drwxr-xr-x  3 root root 4096 dez 17 21:27 5.4.4

```

dmikushin, 2020-01-05:
LGTM:

```
ls -l
total 25604
-rw-rw-r-- 1 marcusmae marcusmae     1197 Jan 19  2019 boot.cmd
-rw-r--r-- 1 root      root          1269 Jan 19  2019 boot.scr
lrwxrwxrwx 1 root      root            37 Sep  4 02:20 dtb -> rk3399-nanopi-m4.dtb_5.3.0-rc7-rk3399
lrwxrwxrwx 1 root      root            22 Sep  4 02:20 Image -> Image_5.3.0-rc7-rk3399
-rw-r--r-- 1 root      root      25029120 Sep  4 02:19 Image_5.3.0-rc7-rk3399
-rw-rw-r-- 1 marcusmae marcusmae  1105188 Sep 24  2017 initrd.img
drwx------ 2 root      root         16384 Sep 25  2017 lost+found
-rw-r--r-- 1 root      root         55139 Sep  4 02:35 rk3399-nanopi-m4.dtb_5.3.0-rc7-rk3399
```

```
ls usr/lib/modules
5.3.0-rc7-rk3399
```

avafinger, 2020-01-05:
I will build 5.4.8 and see what i get... Maybe i test the install and let you know. It takes a few hours.

avafinger, 2020-01-05:
Install success.

```
sudo dpkg -i linux-image-5.4.8_1.0-29.deb
[sudo] password for ubuntu: 
(Reading database ... 67836 files and directories currently installed.)
Preparing to unpack linux-image-5.4.8_1.0-29.deb ...
INFO: Updating Kernel 5.4.4 to 5.4.8
INFO: Updating...
Unpacking linux-image-5.4.8 (1.0-29) over (1.0-29) ...
Setting up linux-image-5.4.8 (1.0-29) ...
OK: Kernel update success! Please reboot with: sync && sudo reboot
```

Uploading...

avafinger, 2020-01-05:
Not really optimized but works (no wifi for this build, next one i will enable it, or you can grab 5.4.8, enable wifi and use this dtb).

```
uname -ra
Linux nanopi-m4 5.4.8 #1 SMP PREEMPT Sun Jan 5 17:52:22 -03 2020 aarch64 aarch64 aarch64 GNU/Linux

```

https://github.com/avafinger/nanopi-m4-ubuntu-base-minimal/releases/tag/v1.33

dmikushin, 2020-01-06:
Dear @avafinger, thank you very much for your hard work. At this point, I can conclude that **one of loader images (idbloader, trust or uboot) of [EOAN release](https://github.com/avafinger/nanopi-m4-ubuntu-base-minimal/releases/tag/v1.27) is broken**. The proof is: I was able to boot EOAN only after rolling the /boot and /rootfs EOAN content on top of the KODI sdcard. You may want to check the EOAN release from scratch, i.e. using the flasher script, not just a kernel.

derMatze1701, 2020-07-15:
@avafinger I can confirm this on my nanopi m4 v1. Image was successfully created with your flasher script to sd card, but does not boot (red lights, sorry I don’t have a debugger to investigate further). Can you please check again? I‘m really looking forward into Ubuntu 19. Thanks for your work!

avafinger, 2020-07-16:
I will build one Image this weekend and check and  possibly fix if something is wrong. There are reports with u-boot problems to boot with HDMI attached. If this is the case, you can boot the board without HDMI cable pluged and then plug the cable after it boots.

Are you trying to boot from eMMC or SD CARD?

derMatze1701, 2020-07-16:
Thank you for looking into it!
HDMI cable is not plugged in, I am trying to boot from SD CARD.

avafinger, 2020-07-16:
There is a new configuration in u-boot for a 2G version, all my builds were tested on a 2G DRAM board.
Do you have a 4GB version?

U-boot 2020.07 still has problems during boot for HDMI output, still not fixed but i will use it.

derMatze1701, 2020-07-16:
I have the 2GB RAM version

avafinger, 2020-07-16:
Can you try the https://github.com/avafinger/nanopi-m4-ubuntu-base-minimal/releases/tag/v1.27 with the latest u-boot (2GB and HDMI disabled).

Untar the attached file and overwrite uboot.img, trust.img and idbloader.img. (make sure they are really the new ones)
It boots fine with SD card here, i just takes 16 or more seconds to display de prompt.

If you see the leds blinking it means the kernel is up, wait a few more seconds to DHCP get the IP (eth0). If you don't see anything on the screen, try to **ssh ubuntu@IP** (find the IP on your router log).

Let me know it this works  for you.

[u-boot.tar.gz](https://github.com/avafinger/nanopi-m4-ubuntu-base-minimal/files/4933869/u-boot.tar.gz)

derMatze1701, 2020-07-16:
I copied the three file, ran the flasher script again, but still red lights.. nothing plugged in except power supply.

derMatze1701, 2020-07-16:
In boot.cmd I can see that env boot_blk is set to 1, so that root=/dev/mmcblk${boot_blk}p2 will become /dev/mmcblk1p2, but  my SDcard is on /dev/mmcblk0 (when booted from eMMC). Is that correct?

avafinger, 2020-07-16:
Booting without eMMC attached and with SD CARD inserted, SD CARD is 1, /dev/mmcblk1

Try first with 0 and 2, since i don't know which is the boot order.

avafinger, 2020-07-16:
if you have eMMC connected, I don't know what boot_from_device value should be, I think you'll need to find out...

derMatze1701, 2020-07-16:
I compared with your older image (https://github.com/avafinger/nanopi-m4-ubuntu-base-minimal#os-image), boot from sd card, emmc attached:

/dev/mmcblk0p1       32768   278527   245760  120M 83 Linux
/dev/mmcblk0p2      278528 15777791 15499264  7.4G 83 Linux

So 0 should be correct

avafinger, 2020-07-16:
Different u-boot, boot order is not the same.

derMatze1701, 2020-07-16:
I now flashed Armbian_20.05.7_Nanopim4_focal_current_5.4.49 to SD Card and it works.
Thanks for your work!

avafinger, 2020-07-20:
Not enough information. Using a new bootloader will fix this.

closed 2020-07-20 by avafinger