avafinger/nanopi-m4-ubuntu-base-minimal #17: Any xhci speed limits on NanoPI?

dmikushin, 2020-01-06:
What seem strange is a USB 3.0 camera won't work on its full FPS, as it does on the desktop, even with the newest EOAN kernel. I have to admit, 5.4 kernel so far shows the best FPS rate of all options, which is a pretty nice job!

FriendlyArm's 4.4 had ugly bugs on xhci, making USB 3.0 mostly not working at all. This was fixed only very recently.

Still, 5.4 kernel is yet far beyond the expected frame rate of desktop. I see CPU load is only about 30%. USB3 device is shown 5000M SuperSpeed, which is correct. The FPS rate seems to get down, if sdcard is intensively in use. Do you know of any measures to improve the xhci performance?

dmikushin, 2020-01-06:
To be more specific, a camera app deploys glib for event/callback handling. So the slowdown could be either due to less frequent event timer clocking on NanoPi, or due to slower xhci "bus" clock itself.

As I said, your kernels give a nice advantage of approx 20%: 500fps instead of 400fps. But it actually should be 800fps.

avafinger, 2020-01-06:
> FriendlyArm's 4.4 had ugly bugs on xhci, making USB 3.0 mostly not working at all. This was fixed only very recently.

For instance, can you point to related fixes on their code?

Could be the way the hardware is implemented?

dmikushin, 2020-01-06:
Sorry, I don't digg into kernel development due to the lack of time. To my experience, FriendlyArm 4.4 was affected at least by [this bug](https://bbs.archlinux.org/viewtopic.php?id=231466), which flooded the syslog, ruining the performance.

> Could be the way the hardware is implemented?

Yes, I guess it could be. On desktops SATA disks and USB controllers usually do not share the same bus. While here, sdcard and usbs are on the same hci. And the eMMC port is hci as well, correct? So they could easily flood each other. One hope might be [PCIe To USB 3.0 X4 HAT for M4](https://www.aliexpress.com/i/33049878123.html), which wires USB3 directly to PCIe. Yet, not clear how PCIe 2x (two lanes!) can do 5Gbs for USB3.0... Yeah, so unfortunately, this board is full of fake sh*t. Do you think other Rockships can do any better?

avafinger, 2020-01-06:
I am not a hardware guy, so not much to add about that. I think for the price Rockchip is the best you can get. I am not sure there is support for the HAT on mainline 5.4 , but 5.5 is almost there, so there could be an improvement with the HAT, but i read you get a higher board temp. anyway, rc5 is out and it is time to give it a try soon...

dmikushin, 2020-01-06:
Do you accept sponsorship to improve certain software problems?

avafinger, 2020-01-06:
I don't think i have the skills and time to improve the kernel. it is just amateur's work. But i can try to help if i can.

dmikushin, 2020-01-06:
Oh, not really the kernel development that much. I do have a lot of housekeeping requests related to our product. Currently your kernel ships a lot of modules, which are definitely no-use. For example, nouveau. I'd be happy to support filtering modules folder to the bare minimum required by this platform.

dmikushin, 2020-01-06:
Another significant usability problem is: nobody could ever understand what's going wrong during early boot without any logging option! I've spend days trying to get uboot or UART to print something without any luck. There must be a clear instruction how to get UART to work. There must be a uboot build to start posting boot status over HDMI as early as possible. I'd support these items right away.

avafinger, 2020-01-06:
>  I've spend days trying to get uboot or UART to print something without any luck

USB-TTL Boot log you get with these parameters:

```
# Machine-generated file - use setup menu in minicom to change parameters.
baudrate         1500000
bits             8
parity           N
stopbits         1

```

The kernel then sets to 115K.

avafinger, 2020-01-06:
I use minicom, configure the port and then fire with:

`sudo minicom -D /dev/ttyUSB0`

dmikushin, 2020-01-06:
> Currently your kernel ships a lot of modules, which are definitely no-use.

I've measured, looks like the base system is ~300MB, but kernel modules bring additional 800MB, which is crazy :)

avafinger, 2020-01-06:
I used RK binaries to embed in u-boot. You might have a look at Armbian which builds atf and u-boot from source, but you need to use their build system.

Here is the script to build from source: https://github.com/armbian/build/blob/master/config/sources/families/rockchip.conf

dmikushin, 2020-01-06:
> I use minicom, configure the port and then fire with:
> 
> `sudo minicom -D /dev/ttyUSB0`

I really try hard to make it work. My UART dongle is [this one](https://www.aliexpress.com/item/32907852614.html?spm=a2g0s.9042311.0.0.27424c4dlThJIn). All is set, but no signal (see screenshot). I'm thinking that I need to re-program the CP2102 to add 1500000 baudrate support as mentioned [here](http://www.olliw.eu/storm32bgc-wiki/How_to_configure_CP2102_USB_adapters_for_high_baud_rates), but the procedure is not clear enough.

[image]

dmikushin, 2020-01-06:
I've checked with ESP32 this UART dongle works well. What chip your UART dongle is based on?

avafinger, 2020-01-06:
Be careful with some UART dongle, i have learned the hard way, some have 3.41v on TX/RX that does not work on NanoPi boards. A long time ago I told FE about the problem while trying to find out why my USB-TTL never worked, and for some other board, it worked. I ended up buying https://www.friendlyarm.com/index.php?route=product/product&product_id=178 ,  problem fixed.

Don't know if it is your case, measure the TX and RX and see what you get.

dmikushin, 2020-01-06:
OK, I got UART for NanoPI M4 to work with [this dongle](https://www.aliexpress.com/item/32772123071.html?spm=a2g0s.9042311.0.0.27424c4d4FI5mm). It's actually an MSP430 programmator, based on a different chip I guess. Not FE's, but works. Thank you very much for your confidence! Is there a way to do shell login through UART?

[image]

avafinger, 2020-01-06:
At the end, it will prompt for the login (UART). You type **ubuntu** / **ubuntu**.
You are in command.

avafinger, 2020-01-06:
```
alex@svn:~$ sudo minicom -D /dev/ttyUSB0
[sudo] password for alex: 
Sorry, try again.
[sudo] password for alex: 

Welcome to minicom 2.7.1

OPTIONS: I18n 
Compiled on Aug 13 2017, 15:25:34.
Port /dev/ttyUSB0, 20:04:51

Press CTRL-A Z for help on special keys

DDR Version 1.17 20190115
In
Channel 0: DDR3, 800MHz
Bus Width=32 Col=10 Bank=8 Row=15 CS=1 Die Bus-Width=16 Size=1024MB
Channel 1: DDR3, 800MHz
Bus Width=32 Col=10 Bank=8 Row=15 CS=1 Die Bus-Width=16 Size=1024MB
256B stride
ch 0 ddrconfig = 0x101, ddrsize = 0x20
ch 1 ddrconfig = 0x101, ddrsize = 0x20
pmugrf_os_reg[2]817281, stride = 0x9
OUT
Boot1: 2018-08-06, version: 1.15
CPUId = 0x0
ChipType = 0x10, mmc: ERROR: SDHCI ERR:cmd:0x102,stat:0x18000
mmc: ERROR: Card did not respond to voltage select!                             
emmc reinit                                                                     
mmc: ERROR: SDHCI ERR:cmd:0x102,stat:0x18000                                    
mmc: ERROR: Card did not respond to voltage select!                             
emmc reinit                                                                     
mmc: ERROR: SDHCI ERR:cmd:0x102,stat:0x18000                                    
mmc: ERROR: Card did not respond to voltage select!                             
SdmmcInit=2 1                                                                   
mmc0:cmd5,20                                                                    
SdmmcInit=0 0                                                                   
BootCapSize=0                                                                   
UserCapSize=30436MB                                                             
FwPartOffset=2000 , 0                                                           
StorageInit ok = 67887                                                          
SecureMode = 0                                                                  
SecureInit read PBA: 0x4                                                        
SecureInit read PBA: 0x404                                                      
SecureInit read PBA: 0x804                                                      
SecureInit read PBA: 0xc04                                                      
SecureInit read PBA: 0x1004                                                     
SecureInit read PBA: 0x1404                                                     
SecureInit read PBA: 0x1804                                                     
SecureInit read PBA: 0x1c04                                                     
SecureInit ret = 0, SecureMode = 0                                              
GPT 0x3190d20 signature is wrong                                                
LoadTrust Addr:0x4000                                                           
No find bl30.bin                                                                
No find bl32.bin                                                                
Load uboot, ReadLba = 2000                                                      
Load OK, addr=0x200000, size=0xa0bf4                                            
RunBL31 0x10000                                                                 
NOTICE:  BL31: v1.3(debug):4f4079b                                             
NOTICE:  BL31: Built : 17:40:23, Oct 11 2018                                    
NOTICE:  BL31: Rockchip release version: v1.1                                   
INFO:    GICv3 with legacy support detected. ARM GICV3 driver initialized in EL3
INFO:    Using opteed sec cpu_context!                                          
INFO:    boot cpu mask: 0                                                       
INFO:    plat_rockchip_pmu_in31: Initializing runtime services                  
WARNING: No OPTEE provided byation. SMC`s destined for OPTEE will return SMC_UNK
ERROR:   Er Preparing for EL3 exit to normal world                              
INFO:    Entry point ad                                                         
                                                                                
[... 381 lines cut ...]
[  OK  ] Started OpenBSD Secure Shell server.                                   
[  OK  ] Started Deferred execution scheduler.                                  
[  OK  ] Started Hostname Service.                                              
[   12.325453] Filesystem uses "xz" compression. This is not supported          
[  OK  ] Stopped Deferred execution scheduler.                                  
         Starting Deferred execution scheduler...                               
[  OK  ] Started Deferred execution scheduler.                                  
[  OK  ] Started Dispatcher daemon for systemd-networkd.                        
[  OK  ] Started Snappy daemon.                                                 
         Starting Wait until snapd is fully seeded...                           
[  OK  ] Started Wait until snapd is fully seeded.                              
[  OK  ] Stopped Deferred execution scheduler.                                  
[FAILED] Failed to start Deferred execution scheduler.                          
See 'systemctl status atd.service' for details.                                 
[  OK  ] Reached target Multi-User System.                                      
[  OK  ] Started Remove Stale Onlin��…ext4 M                                    
[  OK  ] Reached target Graphical Interface.                                    
         Starting Update UTMP about System Runlevel Changes...                  
[  OK  ] Started Update UTMP about System Runlevel Changes.                     
                                                                                
Ubuntu 19.10 nanopi-m4 ttyS2                                                    
                                                                                
nanopi-m4 login: 

```

avafinger, 2020-01-06:
```
ubuntu                                                                          
Password:                                                                       
Last login: Sun Jan  5 23:19:16 UTC 2020 from 192.168.254.253 on pts/0          
Welcome to Ubuntu 19.10 (GNU/Linux 5.4.8 aarch64)                               
                                                                                
 * Documentation:  https://help.ubuntu.com                                      
 * Management:     https://landscape.canonical.com                              
 * Support:ps://ubuntu.com/advantage                                            
                                                                                
 * Overheard at KubeCon: "microk8s.ands#microk8s.status                         
ubuntu@nanopi-m4:~$ 
```

dmikushin, 2020-01-07:
> ## Executing script at 00500000                                                 
> 26055168 bytes read in 1648 ms (15.1 MiB/s)                                     
> 1157052 bytes read in 78 ms (14.1 MiB/s)

Nice eMMC read speed! Which one is that? I can get only 9MB/sec out of mine.

avafinger, 2020-01-07:
it's a 32 GB SD card (Sandisk Ultra) , 80 MB/s read. Vendor info. But many builds i think is wearing out too fast also.

dmikushin, 2020-01-07:
I need to raise ethernet, yet I don't copy most of the kernel modules due to ramdisk economy. The boot log gives me:

```
[   10.169469] dhclient[235]: Error getting hardware address for "eth0": No such device
```

Am I missing some particular kernel module for ethernet support?

TTY login does not start as well, perhaps due to the same missing modules:

```
[ TIME ] Timed out waiting for device /dev/ttyS2.
[   99.565630] systemd[1]: dev-ttyS2.device: Job dev-ttyS2.device/start timed out.
[DEPEND] Dependency failed for Serial Getty on ttyS2.
[   99.589139] systemd[1]: Timed out waiting for device /dev/ttyS2.
[   99.602220] systemd[1]: Dependency failed for Serial Getty on ttyS2.
[  OK  ] Reached target Login Prompts.
[   99.613096] systemd[1]: serial-getty@ttyS2.service: Job serial-getty@ttyS2.service/start failed with result 'dependency'.
[  OK  ] Reached target Multi-User System.
[   99.636980] systemd[1]: dev-ttyS2.device: Job dev-ttyS2.device/start failed with result 'timeout'.
```

Any clue here? Thanks a lot!

dmikushin, 2020-01-07:
I have of course dumped your active module set and brought their .ko files into the rootfs. But this does not help. What am I missing?

This is my log: https://gist.github.com/dmikushin/5d762745573361d0c923d6aacc904ecb

avafinger, 2020-01-07:
Please, attach a complete boot log.

dmikushin, 2020-01-07:
Yes sure, please see here: https://gist.github.com/dmikushin/10bdc259f3bd7f6485861d5cd6183c9c

dmikushin, 2020-01-07:
> [    6.930262] ttyS2 - failed to request DMA

dmikushin, 2020-01-07:
Two more tests:

* I was able to launch it in Qemu. There, the network raises OK, but same issue with TTY.
* Also I have a eMMC-to-sdcard adapter, so I tried to change `boot_from_device` to 1 and boot as from sdcard, but boot does not start (empty UART output).

avafinger, 2020-01-07:
Some tips:

* I see you have a different partition scheme, seems to be only one partition

* You are using U-boot for BSP (Kernel 4.4), for mainline i use: U-Boot 2019.10-rc3-00141-g877294b-dirty (Sep 03 2019)
They switch device ID value, so you may need to adjust boot.cmd in root=/dev/mmcblk**0p2**
You need to change /etc/fstab accordingly, i think you are aware of that.

* For the ttyS2
Do you have something like this in your boot.cmd?
setenv bootargs "console=ttyS2,115200 .....

dmikushin, 2020-01-07:
If I add `console=ttyS2,115200`, then I don't see anything after `Starting kernel ...` in the UART output.

I am booting the the RAM, so there is only one boot/ partition alone, and the boot process should finish at the initrd state, so the root argument is effectively not in use.

dmikushin, 2020-01-07:
At this point, I'm quite certain the cause of the problem is that **initrd cannot have kernel modules**. Don't have an exact pointer, but appears to be it. Am going to test this.

avafinger, 2020-01-07:
> If I add `console=ttyS2,115200`, then I don't see anything after `Starting kernel ...` in the UART output.
> 
> I am booting into the RAM, so there is only one boot/ partition alone, and the boot process should finish at the initrd state, so the root argument is effectively not in use.

You should **not** add this to the boot.cmd!
I think you are rolling your own distro, your UART problem is a known issue and is related to the uart service in the distro, look for this service, if i recall correctly there is a missing symlink or symlink to wrong ttyS*

> initrd cannot have kernel modules
Try to use Built-in modules instead of loadable modules

dmikushin, 2020-01-07:
Unfortunately, I cannot recompile from your release, because source code zip/tar.gz does not contain any source/patches. Do you wish to share them?

dmikushin, 2020-01-07:
I mean do you add anything special on top of https://cdn.kernel.org/pub/linux/kernel/v5.x/linux-5.4.8.tar.gz ?

dmikushin, 2020-01-08:
So what I did now: I created a kernel defconfig based on your kernel config with all modules (=m) removed and all active modules in lsmod added back (=y). The resulting kernel is only 2M larger than original. It boots, yet with non-working UART - I will check this out, thanks for the pointer! And thanks for so kindly leading my way!!

dmikushin, 2020-01-10:
> I think you are rolling your own distro, your UART problem is a known issue and is related to the uart service in the distro, look for this service, if i recall correctly there is a missing symlink or symlink to wrong ttyS*

@avafinger Are you talking about /etc/securetty? I'm trying to find what you've suggested, is it a systemctl service?

dmikushin, 2020-01-10:
I can only see that in stock FriendlyDesktop `ttyS2` is named `ttyFIQ0` instead. But this is not our case.

dmikushin, 2020-01-10:
Maybe you mean `serial-getty@.service` ? But these files are identical for your and mine rootfs.

dmikushin, 2020-01-10:
In kernel logs there is a difference. You kernel:

> ttyS2 - failed to request DMA

My kernel:

> ttyS0 - failed to request DMA

This is out before systemd, so the 0 or 2 number is given by the kernel config??

avafinger, 2020-01-10:
> Maybe you mean `serial-getty@.service` ? But these files are identical for your and mine rootfs.

Yes.

and
ttyS0 = BT
ttyS2 = console

avafinger, 2020-01-10:
Make sure you have:
```

&uart0 {
	pinctrl-names = "default";
	pinctrl-0 = <&uart0_xfer &uart0_rts &uart0_cts>;
	status = "okay";

	bluetooth {
		compatible = "brcm,bcm43438-bt";
		clocks = <&rk808 1>;
		clock-names = "lpo";
		device-wakeup-gpios = <&gpio2 RK_PD2 GPIO_ACTIVE_HIGH>;
		host-wakeup-gpios = <&gpio0 RK_PA4 GPIO_ACTIVE_HIGH>;
		shutdown-gpios = <&gpio0 RK_PB1 GPIO_ACTIVE_HIGH>;
		max-speed = <4000000>;
		pinctrl-names = "default";
		pinctrl-0 = <&bt_reg_on_h &bt_host_wake_l &bt_wake_l>;
		vbat-supply = <&vcc3v3_sys>;
		vddio-supply = <&vcc_1v8>;
	};
};

&uart2 {
	status = "okay";
};
```

dmikushin, 2020-01-11:
Meanwhile, I've tested two xhci measures.

First, wrote a [simple CPU core](https://gist.github.com/dmikushin/4b7aed2765b95a5d03b310b436c110cd) locker, which should bound application threads to individual cores. However, this does not affect the performance or makes it worse, depending on core indexes selection.

Second, I've installed `cpufrequtils` and enforced min/max bounds towards core maximums:
```
sudo apt-get install cpufrequtils
sudo cat /etc/default/cpufrequtils
GOVERNOR="performance"
MIN_SPEED="1500MHz"
MAX_SPEED="2000MHz"
sudo /etc/init.d/cpufrequtils restart
cpufreq-info
```
This also does not improve the performance, yet works in principle: if min/max bounds are lower, then xhci application is slower.

I will also look into how callback threads are waiting for sync. There is a chance that on ARM glib is using a timer with limited resolution, rounding delays towards higher values.

I'd highly appreciate any other ideas!