avafinger/nanopi-m4-ubuntu-base-minimal #19: Provide sources/patches/scripts?

rubenvb, 2020-03-27:
Hi,
I have succesfully used you images on my brand new NanoPi M4.
I have even used just the u-boot parts to start up Archlinux ARM on this board, and everything seems as it should be (meaning: no wifi but everything else seems in working order).

I would love to know how you made the u-boot_v2 files. I have tried to create these but I have no idea which files from the u-boot repository to choose. It would be great if you could provide the exact steps you performed to create your binaries.
If you patch the linux kernel in any way, or use a specific build configuration, could you provide it as well? I'd like to make sure I don't miss anything and can reproduce your great work for my Archlinux setup.

Thanks a bunch for this great work! I hope I can contribute something back some day.

avafinger, 2020-03-28:
Here is the patch for Wifi and HDMI sound output for mainline 5.x
[wifi_hdmi-sound-v2.patch.tar.gz](https://github.com/avafinger/nanopi-m4-ubuntu-base-minimal/files/4397381/wifi_hdmi-sound-v2.patch.tar.gz)

.
As for the u-boot, i have not touched it for a long time, i think i used u-boot-2019.10-rc3 with rockchip blobs following Rockchip instructions. It worked great for 4.4.y but has minor issues with 5.x

rubenvb, 2020-03-31:
Thanks for the quick answer!

I couldn't get your image to work on the eMMC so I wanted to roll the u-boot myself. In the end it turned out that the eMMC device under Linux 5.5 (might be all 5.x?) is actually /dec/mmcblk2, not 0 as your boot.cmd describes. Lost a good day of Corona quarantine on that ;). It all does seem to work, except for the slow boot due to the low BIG CPU core frequency on image decompression. I wonder if I can use an uncompressed image to make that better?

It seems that at least for HDMI output I don't need to patch anything vs Linux 5.5. Wi-Fi isn't high on my list but maybe I'll need a patched kernel anyway if I want hardware accelerated decoding (through V4L2 seems like the best route) in Kodi. I'll take it along when I get to that. Preliminary tests show vanilla Kodi 18.6 running under X11 handling video files fine, but I guess I'll run into CPU limitations there soon...

avafinger, 2020-03-31:
>  /dec/mmcblk2

yeah, that is it, i think i described this in the boot.cmd, cant remember.

>  low BIG CPU core frequency on image decompression

comment on the following lines in rk3399.dtsi and the boot is **fast** again.
**capacity-dmips-mhz = <485>;** 

>  Preliminary tests show vanilla Kodi 18.6 running under X11 handling video files fine

Make sure you have **panfrost** enabled or you are decoding video by software (by software takes a lot of CPUs).
Check out this: https://github.com/avafinger/htop_2.2.2 (if not already)

Good Luck with your quarantine! Good Luck to us all.

rubenvb, 2020-04-01:
> > /dec/mmcblk2
> 
> yeah, that is it, i think i described this in the boot.cmd, cant remember.

All I found was 0 and 1, never a 2 ;).

> > low BIG CPU core frequency on image decompression
> 
> comment on the following lines in rk3399.dtsi and the boot is **fast** again.
> **capacity-dmips-mhz = <485>;**

Well, that change to the dtsi file also has other implications:
https://github.com/torvalds/linux/commit/97df3aa76b4a384e29668f374a8b1034a31aa215
I'd rather figure out a way to properly set the clock speed on the big cores from the bootloader, but I haven't come across anyone mentioning a way to do that except a hack by loading cpufreq a bit earlier:
https://gitlab.collabora.com/rockpi/linux/commit/e55cd521ed67943cc3edcd8b78884ad53741da8c

> > Preliminary tests show vanilla Kodi 18.6 running under X11 handling video files fine
> 
> Make sure you have **panfrost** enabled or you are decoding video by software (by software takes a lot of CPUs).

I verified with glxinfo that X is using panfrost. I doubt it will do much to help video decode performance though. For that it seems I need FFMPEG built with V4L2 request API (and a newer kernel than 5.6 as that one doesn't seem to have the H264 defines needed by FFMPEG).
 
> Check out this: https://github.com/avafinger/htop_2.2.2 (if not already)

Must have been the first thing I did on my nanopi, it's very useful :).
Kernel 5.6 seems to show CPU and GPU temperature when you run `sensors`, which is nice. `lscpu` does not report current frequency yet.

> 
> Good Luck with your quarantine! Good Luck to us all.

Stay healthy.

I'll close this issue now, maybe I'll start a new discussion in a more appropriate place.

closed 2020-04-01 by rubenvb

avafinger, 2020-04-02:
ouch, sorry, you are right. panfrost has nothing to do with hardware decoding. Watch kwiboo/linux-rockchip , I think he is working on rkdec for mainline. :)