avafinger/nanopi-m4-ubuntu-base-minimal #20: Nice work

Askmewho, 2020-07-16:
Hi! I have been testing m4 on mainline woth armbian. I never achieved vpu acceleration while hantro modele It's confirmed to be there. How I need to compile ffmpeg to get hantro to run at least h264 format?? What other formats it can decode?? Thanks on advance.

avafinger, 2020-07-16:
I will try to test VPU this weekend and see what i get. (mairline)

Askmewho, 2020-07-18:
So, the problem seems to be than the fork of ffmpeg thab gives you hw acceleration demands another patches, not just hantro. I am utilizing the most mainline image available for rk3399 on armbian. It seems thab that patch must be done manually. I thought it was already there and it's not, no matter than hantro it's present. So.. I will not have another option than use your images (I mean, I am not that ok with not having focal). I don't know how to apply a patch on armbian build script, but if you know how it would be cool if you inform me how.

avafinger, 2020-07-18:
I am building one IMG today, so you can test it and compare.

Askmewho, 2020-07-18:
Thankss!

avafinger, 2020-07-19:
You can test it: https://github.com/avafinger/nanopi-m4-ubuntu-base-minimal/releases/tag/v1.36

Askmewho, 2020-07-20:
It didn't boot. No problem. I will just install legacy. Thanks for the effort

closed 2020-07-20 by avafinger