avafinger/nanopi-m4-ubuntu-base-minimal #21: Wireguard

lorefx, 2020-07-30:
Hello, thanks again for the big work you're doing.
I'm currently testing the 5.7.9 release with kodi (building it right now) starting from the raw image provided. So far all good.

I know that since kernel 5.6 the wireguard protocol has been included directly in the kernel, but I'm unable to understand how to make it usable on the 5.7.9 kernel provided.

From what I read, I saw that it should be sufficient to apt install wireguard, but it's not the case as it tries to install linux firmware kernel and other things of version 5.3 for raspberry, and obviously I don't want to as it will break the current image

Have you got some clue on how to make wireguard usable?

avafinger, 2020-07-30:
Unfortunately, i have no experience with wireguard.

lorefx, 2020-09-08:
No problem, I'll make my own trial and error compiling. I'll notify to you if successful

closed 2020-09-08 by lorefx