avafinger/nanopi-m4-ubuntu-base-minimal #22: dm-crypt issues, can we get it as a module

nikias, 2020-08-31:
Hi,
when I try to mount a luks volume, I am getting this error:
```
device-mapper: table: 253:0: crypt: unknown target type
device-mapper: reload ioctl on   failed: Invalid argument
```
According to my researches, this bug can be resolved by providing `dm-crypt` as a kernel module instead of built-in.
Could you update the build to reflect that?
That would be pretty dope :)
Thanks

avafinger, 2020-08-31:
Can you tell which kernel?
I think kernel 5.x i have # CONFIG_DM_CRYPT is not set

Would you mind trying 5.9.0-rc3? I am testing it with 20.04 LTS.

nikias, 2020-09-01:
Oh sorry, I was talking about 5.7.9 - if it's not set then obviously it can't work at all. I might give it a shot with 5.9.0-rc3.

avafinger, 2020-09-01:
I have the config below, check if you need anything else:

```
#
# Generic fallback / legacy drivers
#
# CONFIG_PATA_ACPI is not set
# CONFIG_ATA_GENERIC is not set
# CONFIG_PATA_LEGACY is not set
CONFIG_MD=y
CONFIG_BLK_DEV_MD=m
# CONFIG_MD_LINEAR is not set
# CONFIG_MD_RAID0 is not set
# CONFIG_MD_RAID1 is not set
# CONFIG_MD_RAID10 is not set
# CONFIG_MD_RAID456 is not set
# CONFIG_MD_MULTIPATH is not set
# CONFIG_MD_FAULTY is not set
# CONFIG_BCACHE is not set
CONFIG_BLK_DEV_DM_BUILTIN=y
CONFIG_BLK_DEV_DM=m
# CONFIG_DM_DEBUG is not set
# CONFIG_DM_UNSTRIPED is not set
CONFIG_DM_CRYPT=m
# CONFIG_DM_SNAPSHOT is not set
# CONFIG_DM_THIN_PROVISIONING is not set
# CONFIG_DM_CACHE is not set
# CONFIG_DM_WRITECACHE is not set
# CONFIG_DM_EBS is not set
# CONFIG_DM_ERA is not set
# CONFIG_DM_CLONE is not set
CONFIG_DM_MIRROR=m
# CONFIG_DM_LOG_USERSPACE is not set
# CONFIG_DM_RAID is not set
CONFIG_DM_ZERO=m
# CONFIG_DM_MULTIPATH is not set
# CONFIG_DM_DELAY is not set
# CONFIG_DM_DUST is not set
# CONFIG_DM_UEVENT is not set
# CONFIG_DM_FLAKEY is not set
# CONFIG_DM_VERITY is not set
# CONFIG_DM_SWITCH is not set
# CONFIG_DM_LOG_WRITES is not set
# CONFIG_DM_INTEGRITY is not set
# CONFIG_TARGET_CORE is not set
# CONFIG_FUSION is not set

```

avafinger, 2020-09-01:
We have it as a modules now:
```
 LD [M]  drivers/md/dm-crypt.ko
  LD [M]  crypto/essiv.ko

```

avafinger, 2020-09-01:
Please try this one, I did it in a rush and had no time to test but should work.
If you prefer the img file it will take some time.

https://github.com/avafinger/nanopi-m4-ubuntu-base-minimal/releases/tag/v1.37