avafinger/nanopi-m4-ubuntu-base-minimal #23: Kodi on 5.7.x

lorefx, 2020-09-08:
Hi, just a question.
IN the README I see that kodi on the 5.7.x compiles and works, but with limited decoding capabilities.

Can you elaborate more on that? I'd like to understand what it means and what it would be needed to make it work properly as the 4.4 kernel version, or event if it is would be possible in the near future.

Sorry for the "silly" question but I really like to grasp things

avafinger, 2020-09-08:
The problem is not the KODI per se but rkdec in mainline is a WIP, you will get some artifacts while decoding h265 10bit for example. By the time i say this it could have been fixed.
You need to follow the mainline progress and update/patch the kernel to fix things.

Kodi is fine if you run with software decoding for all formats, but you get 300% cpu usage in most cases instead of the ~80% CPU with HW decoding.

lorefx, 2020-10-06:
Hi, just a quick update.

I used the 1.39 release and fresh rebuilt kodi this night. Release 1.4 won't work with my Nanopi M4 v1 2GB (at boot, I see garbage on the screen and it reboots). 1.39 works. Both tested with a 32GB sd card

Kodi rebuilt with the following instructions

`sudo apt install autoconf automake autopoint gettext autotools-dev cmake curl default-jre gawk gcc g++ cpp gdc gperf libasound2-dev libass-dev libavahi-client-dev libavahi-common-dev libbluetooth-dev libbluray-dev libbz2-dev libcdio-dev libcec-dev libp8-platform-dev libcrossguid-dev libcurl4-openssl-dev libcwiid-dev libdbus-1-dev libegl1-mesa-dev libenca-dev libflac-dev libfontconfig1-dev libfmt-dev libfreetype6-dev libfribidi-dev libfstrcmp-dev libgcrypt20-dev libgif-dev libgles2-mesa-dev libglew-dev libglu1-mesa-dev libgnutls28-dev libgpg-error-dev libiso9660-dev libjpeg-dev liblcms2-dev liblirc-dev libltdl-dev liblzo2-dev libmicrohttpd-dev libmysqlclient-dev libnfs-dev libogg-dev libomxil-bellagio-dev libpcre3-dev libplist-dev libpng-dev libpulse-dev libsmbclient-dev libsqlite3-dev libssl-dev libtag1-dev libtiff5-dev libtinyxml-dev libtool libudev-dev libva-dev libvdpau-dev libvorbis-dev libxkbcommon-dev libxmu-dev libxrandr-dev libxslt1-dev libxt-dev netcat lsb-release nasm python-dev python-pil python2-minimal rapidjson-dev swig unzip uuid-dev yasm zip zlib1g-dev libcap-dev libgbm-dev libsndio-dev libdrm-dev libinput-dev libgtest-dev libunistring-dev
sync && sudo reboot

mkdir -p kodi
cd kodi
mkdir -p kodi-build
git clone https://github.com/xbmc/xbmc kodi
cd kodi
sudo make -C tools/depends/target/crossguid PREFIX=/usr
sudo make -C tools/depends/target/flatbuffers PREFIX=/usr
sudo make -C tools/depends/target/libfmt PREFIX=/usr
sudo make -C tools/depends/target/libspdlog PREFIX=/usr
cd ../kodi-build
cmake ../kodi -DCMAKE_INSTALL_PREFIX=/usr/local -DCORE_PLATFORM_NAME=gbm -DGBM_RENDER_SYSTEM=gles
cmake --build . -- VERBOSE=0 -j1
sudo usermod -a -G input,video ubuntu
sudo make install
cd ../kodi
sudo make -j1 -C tools/depends/target/binary-addons PREFIX=/usr/local ADDONS="audiodecoder.* audioencoder.* game.* imagedecoder.* inputstream.* peripheral.* vfs.*"
`

Kodi starts and outputs video, but I had to disable DRMPRIME and HW acceleration otherwise all tested videos would start in audio but no actual frames are drawn on screen. 
Error in kodi log is 
ERROR <general>: CDVDVideoCodecDRMPRIME::AddData - send packet failed: Invalid data found when processing input (-1094995529)

Running videos with HW and drm prime disabled works but CPU is 100% on all cores and some frames are skipped.

If someone is also testing this, maybe we can help each other to solve these issues

avafinger, 2020-10-06:
`Release 1.4 won't work with my Nanopi M4 v1 2GB (at boot, I see garbage on the screen and it reboots). 1.39 works. Both tested with a 32GB sd card`

I think there are 2 types of ddr3 memory used on the NanoPi M4 v1. The garbage you see is expected. My NanoPi M4 version is 2GB 1807 and 4GB 1807 (updated the release info). Can you check yours?

`ERROR : CDVDVideoCodecDRMPRIME::AddData - send packet failed: Invalid data found when processing input (-1094995529)` 

You need a special version of FFmpeg, https://github.com/Kwiboo/FFmpeg

Regarding GPU, check if you have panfrost enabled, panfrost is the vanilla version:

`lsmod|grep panfrost`

Try to grab the boot log using a TTL debug.

lorefx, 2020-10-06:
>  think there are 2 types of ddr3 memory used on the NanoPi M4 v1. The garbage you see is expected. My NanoPi M4 version is 2GB 1807 and 4GB 1807 (updated the release info). Can you check yours?

My nanopi is 
Nanopi M4 (2GB\1GB) 1807

> You need a special version of FFmpeg, https://github.com/Kwiboo/FFmpeg

Ok I'll try that

> Regarding GPU, check if you have panfrost enabled, panfrost is the vanilla version:
lsmod|grep panfrost
Try to grab the boot log using a TTL debug.

ubuntu@nanopi-m4:~$ lsmod | grep panfrost
panfrost               57344  0
gpu_sched          32768  1 panfrost
drm                   577536  9 gpu_sched,drm_kms_helper,dw_mipi_dsi,rockchipdrm,dw_hdmi,panfrost,analogix_dp

And this is my boot log

[    0.000000] Booting Linux on physical CPU 0x0000000000 [0x410fd034]
[    0.000000] Linux version 5.9.0-rc7 (alex@svn) (aarch64-linux-gnu-gcc (Linaro GCC 7.5-2019.12-rc1) 7.5.0, GNU ld (Linaro_Binutils-2019.12-rc1) 2.28.2.20170706) #1 SMP PREEMPT Tue Sep 29 18:59:14 -03 2020
[    0.000000] Machine model: FriendlyElec NanoPi M4
[    0.000000] efi: UEFI not found.
[    0.000000] cma: Reserved 128 MiB at 0x0000000071c00000
[    0.000000] NUMA: No NUMA configuration found
[    0.000000] NUMA: Faking a node at [mem 0x0000000000200000-0x000000007fffffff]
[    0.000000] NUMA: NODE_DATA [mem 0x7fbac100-0x7fbadfff]
[    0.000000] Zone ranges:
[    0.000000]   DMA      [mem 0x0000000000200000-0x000000003fffffff]
[    0.000000]   DMA32    [mem 0x0000000040000000-0x000000007fffffff]
[    0.000000]   Normal   empty
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000200000-0x000000007fffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000200000-0x000000007fffffff]
[    0.000000] On node 0 totalpages: 523776
[    0.000000]   DMA zone: 4088 pages used for memmap
[    0.000000]   DMA zone: 0 pages reserved
[    0.000000]   DMA zone: 261632 pages, LIFO batch:63
[    0.000000]   DMA32 zone: 4096 pages used for memmap
[    0.000000]   DMA32 zone: 262144 pages, LIFO batch:63
[    0.000000] psci: probing for conduit method from DT.
[    0.000000] psci: PSCIv1.0 detected in firmware.
[    0.000000] psci: Using standard PSCI v0.2 function IDs
[    0.000000] psci: MIGRATE_INFO_TYPE not supported.
[    0.000000] psci: SMC Calling Convention v1.0
[    0.000000] percpu: Embedded 22 pages/cpu s52888 r8192 d29032 u90112
[    0.000000] pcpu-alloc: s52888 r8192 d29032 u90112 alloc=22*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 [0] 4 [0] 5 
[    0.000000] Detected VIPT I-cache on CPU0
[    0.000000] CPU features: detected: ARM erratum 845719
[    0.000000] CPU features: detected: GIC system register CPU interface
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 515592
[    0.000000] Policy zone: DMA32
[    0.000000] Kernel command line: earlyprintk root=/dev/mmcblk1p2 rw rootfstype=ext4 rootwait fsck.repair=yes panic=10 no_console_suspend consoleblank=0
[    0.000000] Dentry cache hash table entries: 262144 (order: 9, 2097152 bytes, linear)
[    0.000000] Inode-cache hash table entries: 131072 (order: 8, 1048576 bytes, linear)
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] software IO TLB: mapped [mem 0x3bfff000-0x3ffff000] (64MB)
[    0.000000] Memory: 1835440K/2095104K available (12156K kernel code, 780K rwdata, 4604K rodata, 2496K init, 506K bss, 128592K reserved, 131072K cma-reserved)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=6, Nodes=1
[    0.000000] rcu: Preemptible hierarchical RCU implementation.
[    0.000000] rcu: 	RCU restricting CPUs from NR_CPUS=8 to nr_cpu_ids=6.
[    0.000000] 	Trampoline variant of Tasks RCU enabled.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 25 jiffies.
[    0.000000] rcu: Adjusting geometry for rcu_fanout_leaf=16, nr_cpu_ids=6
[    0.000000] NR_IRQS: 64, nr_irqs: 64, preallocated irqs: 0
[    0.000000] GICv3: GIC: Using split EOI/Deactivate mode
[    0.000000] GICv3: 256 SPIs implemented
[... 526 lines cut ...]
[    9.422596] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac4356-sdio for chip BCM4356/2
[    9.545824] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac4356-sdio for chip BCM4356/2
[    9.547709] brcmfmac: brcmf_c_process_clm_blob: no clm_blob available (err=-11), device may have limited channels available
[    9.548292] brcmfmac: brcmf_c_preinit_dcmds: Firmware: BCM4356/2 wl0: Apr 18 2017 13:54:43 version 7.35.184.1.r5 (A2 Station/P2P) FWID 01-efd4ca0d
[    9.653132] rockchip_vdec: module is from the staging directory, the quality is unknown, you have been warned.
[    9.654785] rkvdec ff660000.video-codec: Adding to iommu group 1
[    9.666754] asoc-simple-card hdmi-sound: ASoC: no DMI vendor name!
[    9.895225] hantro_vpu: module is from the staging directory, the quality is unknown, you have been warned.
[    9.898461] hantro-vpu ff650000.video-codec: Adding to iommu group 0
[    9.899031] hantro-vpu ff650000.video-codec: registered rockchip,rk3399-vpu-enc as /dev/video1
[    9.899128] hantro-vpu ff650000.video-codec: registered rockchip,rk3399-vpu-dec as /dev/video2
[   10.051286] EXT4-fs (mmcblk1p1): mounted filesystem with ordered data mode. Opts: (null)
[   10.051311] ext4 filesystem being mounted at /boot supports timestamps until 2038 (0x7fffffff)
[... cut ...]
[   11.026553] zram0: detected capacity change from 0 to 168120320
[   11.240574] Adding 164176k swap on /dev/zram0.  Priority:5 extents:1 across:164176k SS
[   11.468901] Bluetooth: BNEP (Ethernet Emulation) ver 1.3
[   11.468904] Bluetooth: BNEP filters: protocol multicast
[   11.468913] Bluetooth: BNEP socket layer initialized
[   13.445890] rk_gmac-dwmac fe300000.ethernet eth0: Link is Up - 1Gbps/Full - flow control off
[   13.445942] IPv6: ADDRCONF(NETDEV_CHANGE): eth0: link becomes ready
[   33.768713] vcca0v9_s3: disabling

avafinger, 2020-10-06:
The boot log should be from the one that does not boot, just to have a look and see why it hangs.

Can you run this test and report the output/result?:

        sudo ./drmgl

[drmgl.tar.gz](https://github.com/avafinger/nanopi-m4-ubuntu-base-minimal/files/5334317/drmgl.tar.gz)

lorefx, 2020-10-07:
> Try to grab the boot log using a TTL debug.

Regarding the non-booting image, I don't have a serial cable. Maybe I can let it boot and then recover directly from the sd the journal log? Is there any other mean to provide you the log?

> Can you run this test and report the output/result?:
    sudo ./drmgl

sudo ./drmgl        
[sudo] password for ubuntu: 
no connected connector!
failed to initialize DRM

> You need a special version of FFmpeg, https://github.com/Kwiboo/FFmpeg

I cloned the default branch (https://github.com/Kwiboo/FFmpeg/tree/v4l2-request-hwaccel-4.3) and tried to compile as
`./configure` --prefix=/usr --enable-nonfree --enable-gpl --enable-version3 --enable-libmp3lame --enable-libpulse --enable-libv4l2 --enable-v4l2-request --enable-v4l2-m2m --enable-libudev --enable-libdrm --enable-shared
make -j2
'

and Ithe compilation stops with

libavcodec/v4l2_request.c:` In function 'v4l2_request_queue_decode':
libavcodec/v4l2_request.c:284:90: error: 'V4L2_BUF_FLAG_M2M_HOLD_CAPTURE_BUF' undeclared (first use in this function)
  284 | l2_request_queue_buffer(ctx, req->request_fd, &req->output, last_slice ? 0 : V4L2_BUF_FLAG_M2M_HOLD_CAPTURE_BUF);
      |                                                                              ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~

libavcodec/v4l2_request.c:284:90: note: each undeclared identifier is reported only once for each function it appears in
libavcodec/v4l2_request.c: In function 'ff_v4l2_request_decode_slice':
libavcodec/v4l2_request.c:368:37: error: 'V4L2_BUF_CAP_SUPPORTS_M2M_HOLD_CAPTURE_BUF' undeclared (first use in this function)
  368 |     if ((req->output.capabilities & V4L2_BUF_CAP_SUPPORTS_M2M_HOLD_CAPTURE_BUF) != V4L2_BUF_CAP_SUPPORTS_M2M_HOLD_CAPTURE_BUF)
      |                                     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
make: *** [ffbuild/common.mak:59: libavcodec/v4l2_request.o] Error 1`

This makes me think that the kernel "maybe" is not providing the V4L2_BUF_CAP_SUPPORTS_M2M_HOLD_CAPTURE_BUF flag. Have you any advice on this?

I used the 1.39 vanilla release, made an apt update && upgrade and nothing else

avafinger, 2020-10-07:
> no connected connector!

**Connectors**

"A connector is the final destination of pixel-data on a device, and usually connects directly to an external display device like a monitor or laptop panel. A connector can only be attached to one encoder at a time. The connector is also the structure where information about the attached display is kept, so it contains fields for display data, EDID data, DPMS and connection status, and information about modes supported on the attached displays."

Really weird, maybe you have issues with the LCD display in use? Do You have HDMI connection (attached) and what is the resolution? Some unusual  EDID information on a lower level I would think.  

`for p in /sys/class/drm/*/status; do con=${p%/status}; echo -n "${con#*/card?-}: "; cat $p; done`
Let's check what is connected (in my case):
**HDMI-A-1: connected**

```
alex@svn:~$ ssh ubuntu@192.168.254.49
ubuntu@192.168.254.49's password: 
Welcome to Ubuntu 20.04.1 LTS (GNU/Linux 5.9.0-rc7 aarch64)

 * Documentation:  https://help.ubuntu.com
 * Management:     https://landscape.canonical.com
 * Support:        https://ubuntu.com/advantage
Last login: Tue Oct  6 19:56:36 2020
ubuntu@nanopi-m4:~$ for p in /sys/class/drm/*/status; do con=${p%/status}; echo -n "${con#*/card?-}: "; cat $p; done
HDMI-A-1: connected
ubuntu@nanopi-m4:~$ for i in $(ls -1 /sys/class/drm/*/modes); do echo "$i:"; cat $i; done
/sys/class/drm/card1-HDMI-A-1/modes:
1920x1080
1920x1080
1920x1080i
1920x1080i
1920x1080
1920x1080i
1920x1080
1920x1080
1920x1080
1280x1024
1280x720
1280x720
1280x720
1280x720
1024x768
800x600
720x576
720x480
720x480
ubuntu@nanopi-m4:~$ 
```

**Also:**

```
ubuntu@nanopi-m4:~$ dmesg | grep -i drm
[    7.240277] systemd[1]: Starting Load Kernel Module drm...
[    8.811073] panfrost ff9a0000.gpu: [drm:panfrost_devfreq_init [panfrost]] Failed to register cooling device
[    8.811359] [drm] Initialized panfrost 1.1.0 20180908 for ff9a0000.gpu on minor 0
[    8.888904] rockchip-drm display-subsystem: bound ff8f0000.vop (ops vop_component_ops [rockchipdrm])
[    8.889007] [drm] unsupported AFBC format[3231564e]
[    8.890543] rockchip-drm display-subsystem: bound ff900000.vop (ops vop_component_ops [rockchipdrm])
[    8.891332] rockchip-drm display-subsystem: bound ff940000.hdmi (ops dw_hdmi_rockchip_ops [rockchipdrm])
[    9.015120] rockchip-drm display-subsystem: [drm] fb0: rockchipdrmfb frame buffer device
[    9.034499] [drm] Initialized rockchip 1.0.0 20140818 for display-subsystem on minor 1
ubuntu@nanopi-m4:~$ 

```

**and**

```
ubuntu@nanopi-m4:~$ dmesg | grep Linux
[    0.000000] Booting Linux on physical CPU 0x0000000000 [0x410fd034]
[    0.000000] Linux version 5.9.0-rc7 (alex@svn) (aarch64-linux-gnu-gcc (Linaro GCC 7.5-2019.12-rc1) 7.5.0, GNU ld (Linaro_Binutils-2019.12-rc1) 2.28.2.20170706) #1 SMP PREEMPT Tue Sep 29 18:59:14 -03 2020
[... 134 lines cut ...]
    Maximum TMDS clock: 150 MHz
Detailed mode: Clock 74.250 MHz, 800 mm x 450 mm
               1920 2008 2052 2200 ( 88  44 148)
                540  542  547  562 (  2   5  15)
               +hsync +vsync
               VertFreq: 60.000i Hz, HorFreq: 33.750 kHz
Detailed mode: Clock 74.250 MHz, 800 mm x 450 mm
               1280 1390 1430 1650 (110  40 220)
                720  725  730  750 (  5   5  20)
               +hsync +vsync
               VertFreq: 60.000 Hz, HorFreq: 45.000 kHz
Detailed mode: Clock 148.500 MHz, 160 mm x 90 mm
               1920 2008 2052 2200 ( 88  44 148)
               1080 1084 1089 1125 (  4   5  36)
               +hsync +vsync
               VertFreq: 60.000 Hz, HorFreq: 67.500 kHz
Detailed mode: Clock 74.250 MHz, 708 mm x 398 mm
               1280 1720 1760 1980 (440  40 220)
                720  725  730  750 (  5   5  20)
               +hsync +vsync
               VertFreq: 50.000 Hz, HorFreq: 37.500 kHz
Checksum: 0x15
ubuntu@nanopi-m4:~$ 

```

avafinger, 2020-10-07:
and finally, get current fbdev settings:

```
buntu@nanopi-m4:~$ sudo apt-get install fbset
Reading package lists... Done
Building dependency tree       
Reading state information... Done
The following NEW packages will be installed:
  fbset
0 upgraded, 1 newly installed, 0 to remove and 4 not upgraded.
Need to get 117 kB of archives.
After this operation, 314 kB of additional disk space will be used.
Get:1 http://ports.ubuntu.com/ubuntu-ports focal/main arm64 fbset arm64 2.1-31 [117 kB]
Fetched 117 kB in 1s (79.6 kB/s)
Selecting previously unselected package fbset.
(Reading database ... 37354 files and directories currently installed.)
Preparing to unpack .../fbset_2.1-31_arm64.deb ...
Unpacking fbset (2.1-31) ...
Setting up fbset (2.1-31) ...
Processing triggers for man-db (2.9.1-1) ...
ubuntu@nanopi-m4:~$ fbset

mode "1920x1080"
    geometry 1920 1080 1920 1080 32
    timings 0 0 0 0 0 0 0
    accel true
    rgba 8/16,8/8,8/0,0/0
endmode

ubuntu@nanopi-m4:~$ 

```

lorefx, 2020-10-07:
> Really weird, maybe you have issues with the LCD display in use? Do You have HDMI connection (attached) and what is the resolution? Some unusual EDID information on a lower level I would think.
>
> for [...]

I'm currently using an hdmi connected to my full hd tv. Always used the same cable and tv, even with all other versions based on the 4.4 kernel

Output:

```
ubuntu@nanopi-m4:~$ for p in /sys/class/drm/*/status; do con=${p%/status}; echo -n "${con#*/card?-}: "; cat $p; done
HDMI-A-1: connected

ubuntu@nanopi-m4:~$ for i in $(ls -1 /sys/class/drm/*/modes); do echo "$i:"; cat $i; done
/sys/class/drm/card1-HDMI-A-1/modes:
1920x1080
1920x1080
1920x1080i
1920x1080
1920x1080
1920x1080i
1920x1080
1920x1080
1920x1080
2880x576
2880x576i
2880x480
2880x480
1280x1024
1152x864
1280x720
1280x720
1280x720
1440x576
1024x768
1440x480
1440x480
800x600
720x576
720x480
720x480

ubuntu@nanopi-m4:~$  dmesg | grep -i drm
[    5.009420] systemd[1]: Starting Load Kernel Module drm...
[    8.991176] panfrost ff9a0000.gpu: [drm:panfrost_devfreq_init [panfrost]] Failed to register cooling device
[    8.991493] [drm] Initialized panfrost 1.1.0 20180908 for ff9a0000.gpu on minor 0
[    9.314008] rockchip-drm display-subsystem: bound ff8f0000.vop (ops vop_component_ops [rockchipdrm])
[    9.314046] [drm] unsupported AFBC format[3231564e]
[    9.315312] rockchip-drm display-subsystem: bound ff900000.vop (ops vop_component_ops [rockchipdrm])
[    9.315916] rockchip-drm display-subsystem: bound ff940000.hdmi (ops dw_hdmi_rockchip_ops [rockchipdrm])
[    9.436541] rockchip-drm display-subsystem: [drm] fb0: rockchipdrmfb frame buffer device
[    9.450408] [drm] Initialized rockchip 1.0.0 20140818 for display-subsystem on minor 1

ubuntu@nanopi-m4:~$ dmesg | grep Linux
[    0.000000] Booting Linux on physical CPU 0x0000000000 [0x410fd034]
[    0.000000] Linux version 5.9.0-rc7 (alex@svn) (aarch64-linux-gnu-gcc (Linaro GCC 7.5-2019.12-rc1) 7.5.0, GNU ld (Linaro_Binutils-2019.12-rc1) 2.28.2.20170706) #1 SMP PREEMPT Tue Sep 29 18:59:14 -03 2020
[    0.733113] pps_core: LinuxPPS API ver. 1 registered
[    0.739501] Advanced Linux Sound Architecture Driver Initialized.
[    1.901461] usb usb1: Manufacturer: Linux 5.9.0-rc7 ehci_hcd
[    1.937248] usb usb2: Manufacturer: Linux 5.9.0-rc7 ehci_hcd
[    2.017070] usb usb3: Manufacturer: Linux 5.9.0-rc7 ohci_hcd
[    2.092837] usb usb4: Manufacturer: Linux 5.9.0-rc7 ohci_hcd
[    2.112359] usb usb5: Manufacturer: Linux 5.9.0-rc7 xhci-hcd
[    2.125878] usb usb6: Manufacturer: Linux 5.9.0-rc7 xhci-hcd
[    2.145087] usb usb7: Manufacturer: Linux 5.9.0-rc7 xhci-hcd
[    2.158550] usb usb8: Manufacturer: Linux 5.9.0-rc7 xhci-hcd
[    6.924890] mc: Linux media interface: v0.10
[    7.633373] videodev: Linux video capture interface: v2.00

ubuntu@nanopi-m4:~$ sudo apt-get install edid-decode
[sudo] password for ubuntu: 
[... 168 lines cut ...]
Building dependency tree       
Reading state information... Done
The following NEW packages will be installed:
  fbset
0 upgraded, 1 newly installed, 0 to remove and 0 not upgraded.
Need to get 117 kB of archives.
After this operation, 314 kB of additional disk space will be used.
Get:1 http://ports.ubuntu.com/ubuntu-ports focal/main arm64 fbset arm64 2.1-31 [117 kB]
Fetched 117 kB in 0s (534 kB/s) 
Selecting previously unselected package fbset.
(Reading database ... 49310 files and directories currently installed.)
Preparing to unpack .../fbset_2.1-31_arm64.deb ...
Unpacking fbset (2.1-31) ...
Setting up fbset (2.1-31) ...
Processing triggers for man-db (2.9.1-1) ...
ubuntu@nanopi-m4:~$ fbset

mode "1920x1080"
    geometry 1920 1080 1920 1080 32
    timings 0 0 0 0 0 0 0
    accel true
    rgba 8/16,8/8,8/0,0/0
endmode

```

lorefx, 2020-10-07:
Ok wait ... I don't know what I did but maybe turning on the M4 after the aplifier/tv changed something
Now i see

`ubuntu@nanopi-m4:~$ sudo ./drmgl
Using display 0xaaaad7e8dc10 with EGL version 1.4
EGL Version "1.4"
EGL Vendor "Mesa Project"
EGL Extensions "EGL_ANDROID_blob_cache EGL_EXT_buffer_age EGL_EXT_image_dma_buf_import EGL_KHR_cl_event2 EGL_KHR_config_attribs EGL_KHR_create_context EGL_KHR_create_context_no_error EGL_KHR_fence_sync EGL_KHR_get_all_proc_addresses EGL_KHR_gl_colorspace EGL_KHR_gl_renderbuffer_image EGL_KHR_gl_texture_2D_image EGL_KHR_gl_texture_3D_image EGL_KHR_gl_texture_cubemap_image EGL_KHR_image EGL_KHR_image_base EGL_KHR_image_pixmap EGL_KHR_no_config_context EGL_KHR_partial_update EGL_KHR_reusable_sync EGL_KHR_surfaceless_context EGL_EXT_pixel_format_float EGL_KHR_wait_sync EGL_MESA_configless_context EGL_MESA_drm_image EGL_MESA_image_dma_buf_export EGL_MESA_query_driver EGL_WL_bind_wayland_display "
GL Extensions: "GL_EXT_blend_minmax GL_EXT_multi_draw_arrays GL_EXT_texture_format_BGRA8888 GL_OES_compressed_ETC1_RGB8_texture GL_OES_depth24 GL_OES_element_index_uint GL_OES_fbo_render_mipmap GL_OES_mapbuffer GL_OES_rgb8_rgba8 GL_OES_stencil8 GL_OES_texture_3D GL_OES_texture_float GL_OES_texture_float_linear GL_OES_texture_half_float GL_OES_texture_half_float_linear GL_OES_texture_npot GL_OES_vertex_half_float GL_OES_EGL_image GL_OES_depth_texture GL_OES_packed_depth_stencil GL_EXT_texture_type_2_10_10_10_REV GL_OES_get_program_binary GL_APPLE_texture_max_level GL_EXT_discard_framebuffer GL_EXT_read_format_bgra GL_EXT_frag_depth GL_NV_fbo_color_attachments GL_OES_EGL_image_external GL_OES_EGL_sync GL_OES_vertex_array_object GL_EXT_occlusion_query_boolean GL_EXT_texture_rg GL_EXT_unpack_subimage GL_NV_draw_buffers GL_NV_read_buffer GL_NV_read_depth GL_NV_read_depth_stencil GL_NV_read_stencil GL_EXT_draw_buffers GL_EXT_map_buffer_range GL_KHR_debug GL_KHR_texture_compression_astc_ldr GL_OES_required_internalformat GL_OES_surfaceless_context GL_EXT_separate_shader_objects GL_EXT_compressed_ETC1_RGB8_sub_texture GL_EXT_draw_elements_base_vertex GL_EXT_texture_border_clamp GL_KHR_context_flush_control GL_OES_draw_elements_base_vertex GL_OES_texture_border_clamp GL_KHR_no_error GL_KHR_texture_compression_astc_sliced_3d GL_KHR_parallel_shader_compile "
failed to set mode: Permission denied`

lorefx, 2020-10-07:
**_SOLVED_**: after adding the user to the **render** group it worked as a charm