avafinger/nanopi-m4-ubuntu-base-minimal #25: v1 - "reboot

mikegleasonjr, 2020-09-28:


avafinger, 2020-09-28:
The fix is here: https://github.com/avafinger/nanopi-m4-ubuntu-base-minimal#mainline-u-boot

avafinger, 2020-09-28:
See these interesting findings
Yesterday i made an Image with 5.9.0-rc6 with a new SD CARD (high speed 32 GB) while I still have one SD CARD with 8GB with the fix for the reboot using the same kernel.

With the 8GB SD CARD, i have no problems rebooting many times.
With the 32 GB SD CARD, i can see a new issue with u-boot:

Rebooting with 32 GB SD CARD:
```
U-Boot TPL 2020.10-rc3-00012-g9f04a634ef (Aug 29 2020 - 23:09:21)               
Channel 0: DDR3, 933MHz                                                         
BW=32 Col=10 Bk=8 CS0 Row=15 CS=1 Die BW=16 Size=1024MB                         
Channel 1: DDR3, 933MHz                                                         
BW=32 Col=10 Bk=8 CS0 Row=15 CS=1 Die BW=16 Size=102456B stride                 
Trying to boot from BOOTROM                                                     
Returning to boot ROM...                                                        
                                                                                
U-Boot SPL 2020.10-rc3-00012-g9f04a634ef (Aug 29 2020 - 23:09:00)               
Trying to boot from MMC1                                                        
mmc_load_image_raw_sector: mmc block read error                                 
Trying to boot from MMC2                                                        
mmc_load_image_raw_sector: mmc block read error                                 
Trying to boot from MMC1                                                        
mmc_load_image_raw_sector: mmc block read error                                 
SPL: failed to boot from all boot devices                                       
### ERROR ### 

```

Rebooting with 8 GB:
```
U-Boot TPL 2020.10-rc3-00012-g9f04a634ef (Aug 29 2020 - 23:09:21)               
Channel 0: DDR3, 933MHz                                                         
BW=32 Col=10 Bk=8 CS0 Row=15 CS=1 Die BW=16 Size=1024MB                         
Channel 1: DDR3, 933MHz                                                         
BW=32 Col=10 Bk=8 CS0 Row=15 CS=1 Die BW=16 Size=1024MB                         
256B stride                                                                     
Trying to boot from BOOTROM                                                     
Returning to boot ROM...                                                        
                                                                                
U-Boot SPL 2020.10-rc3-00012-g9f04a634ef (Aug 29 2020 - 23:09:21 +0000)         
Trying to boot from MMC1                                                        
                                                                                
                                                                                
U-Boot 2020.10-rc3-00012-g9f04a634ef (Aug 29 2020 - 23:09:21                    
                                                            Model: FriendlyElec4
2 GiB                                                                           
PMIC:  RK808                                                                    
MMC:   mmc@fe310000: 2, mmc@fe320000: 1, sdhci@fe330000: 0                      
Loading Environment from MMC... Card did not respond to voltage select!         
*** Warning - No block device, using default environment                        
                                                                                
In:    serial                                                                   
Out:   serial                                                                   
Err:   serial                                                                   
Model: FriendlyElec NanoPi M4           
```

I can vaguely remember some discussions about issues with SD CARD voltage for some SD CARDs models, the kernel I used is vanilla 5.9.0-rc6 and now 5.9.0-rc7.  Maybe this is the issue above, don't know if the patch is applied, or this patch is upstream now and is the one to blame. I think Armbian had a thread about it.

avafinger, 2020-09-28:
Ok, i think i found a workaround for this.
I will build a new Image and in case you still have problems with reboot you can use this Image.

avafinger, 2020-09-30:
I had to rebuild the Image with old gcc, gcc-10 gives some weird results.
I have tested it in all my SD CARDS with success. 
Try this and give some feedback: https://github.com/avafinger/nanopi-m4-ubuntu-base-minimal/releases/tag/v1.39

mikegleasonjr, 2020-09-30:
Thanks, I will try it... I have the 4GB version can I still use it?

avafinger, 2020-09-30:
Humm,not with this Image, it is for 8GB minimum, I will provide a way to install it if you have a Linux box. But you can mount the 8GB in your Linux box e resize it to 4 GB, but the Image after installed occupies 3.5 GB due to gcc and dev tools.
You can always mount it, chroot to it, and remove the páckages to free additional space.

mikegleasonjr, 2020-09-30:
No I meant the 4GB RAM version :)

avafinger, 2020-09-30:
No boot on 4 GB, just need the u-boot for 4GB version. I will build one.

avafinger, 2020-09-30:
I don't know why it does not initialize DRAM on the 4GB. I have compiled the u-boot 4G version that loads lpddr 4G but get error on DRAm init.

Attached is a working bootloader for 2GB and 4GB DRAM (nanopi m4 v1) If you have a Linux box you can flash the 2G Image file and write the new bootloader. I thought the 4GB version of the u-boot was for DDR4 but it loads lpddr3 timings.

But anyway, the shutdown and reboot work for both, 2GB and 4GB NanoPi M4 v1.

## instructions
* Download the attached file and unzip in your PC (Linux) or if you have a board (any board) where you can mount the SD CARD (the 2GB Image file) or just insert into the free sd card slot and you can write the bootloader to the SD CARD using the command:

* find the mmcblk or sd block where the SD CARD is,
* write the boot loader as the instructions

		sudo dd if=./idbloader.img of=/dev/sdX seek=64 conv=notrunc
		sync
		sudo dd if=./u-boot/trust.img of=/dev/sdX seek=24576 conv=notrunc
		sync
		sudo dd if=./uboot.img of=/dev/sdX seek=16384 conv=notrunc
		sync
		
		where X is the block letter from 1 to 3
		
		
		sudo dd if=./idbloader.img of=/dev/mmcblkY seek=64 conv=notrunc
		sync
		sudo dd if=./u-boot/trust.img of=/dev/mmcblkY seek=24576 conv=notrunc
		sync
		sudo dd if=./uboot.img of=/dev/mmcblkY seek=16384 conv=notrunc
		sync
		
		where Y is the block number from 1 to 3
	
	
use **df -lh** to find out where the SD CARD is.
[u-boot.tar.gz](https://github.com/avafinger/nanopi-m4-ubuntu-base-minimal/files/5307493/u-boot.tar.gz)

avafinger, 2020-09-30:
Boot log from a 4GB NanoPi M4 v1:
	MR11=0x1F                                                                
	MR12=0x1F                                                                       
	MR13=0x1F                                                                       
	MR14=0x1F                                                                       
	MR15=0x1F                                                                       
	MR16=0x1                                                                        
	MR3=0x58                                                                        
	MR4=0x2                                                                         
	MR5=0x1                                                                         
	MR6=0x5                                                                         
	MR7=0x0                                                                         
	MR8=0x1F                                                                        
	MR9=0x13=0x1F                                                                   
	MR14=0x1F                                                                       
	MR15=0x1F                                                                       
	MR16=0x1F                                                                       
	Bus Width=32 Col=10 Bas-Width=32 Size=2048MB                                    
	Channel 1: LPDDR3, 800MHz                                                       
	CS = 0                                                                          
	MR0=0                                                                           
	MR5=0x1                                                                         
	MR6=0x5                                                                         
	MR7=0x0                                                                         
	MR8=0x1F                                                                        
	MR9=0x1F                                                                        
	MR10=0x1F                                                                       
	MR11                                                                            
	MR15=0x1F                                                                       
	MR16=0x1F                                                                       
	CS = 1                                                                          
	MR0=0x58                                                                        
	MR1=0x58                                                                        
	MR2=0x58                                                                        
	MRx0                                                                            
	MR8=0x1F                                                                        
	MR9=0x1F                                                                        
	MR10=0x1F                                                                       
	MR11=0x1F                                                                       
	MR12=0x1F                                                                       
	MR13=0xBus Width=32 Col=10 Bank=8 Row=15/15 CS=2 Die Bus-Width=32 Size=ch 0 ddr0
	ch 1 ddrconfig = 0x101, ddrsize = 0x2020                                        
	Boot1 Release Time: Dec 24 2019 18:00:26, version: 1.24                         
	CPUId = 0x0                                                                     
	ChipType = 0x10, 333                                                            
	mmc: ERROR: SDHCI ERR:cmd:0x102,stat:0x18000                                    
	mmc: ERROR: Card did not respond to voltage select!                             
	emmc reinit                                                                     
	mmc: ERROR: SDHCI ERR:cmd:0x102,stat:0x18000                                    
	mmc: ERROR: Card did not respond tmmc: ERROR: SDHCI ERR:cmd:0x102,stat:0x18000  
	mmc: ERROR: Card d                                                              
	mmc0:cmd5,20                                                                    
	SdmmcInit=0 0                                                                   
	BootCapSize=0                                                                   
	UserCapSize=7704MB                                                              
	FwPartOffset=2000 , 0                                                           
	StorageInit ok = 45301                                                          
	SecureMode = 0                                                                  
	SecureInit read PBA: 0x4                                                        
	SecureInit read PBA: 0x404                                                      
	SecureInit read PBA: 0x804                                                      
	SecureInit read PBA: 0xc04                                                      
	SecureInit read PBA: 0x1004                                                     
	SecureInit read PBA: 0x1404                                                     
	SecureInit read PBA: 0x1804                                                     
	SecureInit read PBA: 0x1c04                                                     
	SecureInit ret = 0, SecureMode = 0                                              
[... cut ...]
	                                                                                
	SoC: Rockchip rk3399                                                            
	ResModel: FriendlyElec NanoPi M4                                                
	DRAM:  3.9 GiB                                                                  
	PMIC:  RK808                                                                    
	MMC:   mmc@fe310000: 2, mmc@fe320000: 1, sdhci@fe330000: 0                      
	Loading Environment from MMC... Card did not respond to voltage select!

avafinger, 2020-09-30:
Here is the Image file:
https://github.com/avafinger/nanopi-m4-ubuntu-base-minimal/releases/tag/v1.40

avafinger, 2020-10-01:
Did it work for you?

mikegleasonjr, 2020-10-04:
Hi, thank you very much for all the work, this is really appreciated! Will try tonight as I haven't had the time yet to try.

Thanks again!