avafinger/nanopi-m4-ubuntu-base-minimal #28: rt5651-sound

razum2um, 2020-11-12:
I've built mainline 5.9 kernel and have only hdmi audio

```
# aplay -l
**** List of PLAYBACK Hardware Devices ****
card 0: hdmisound [hdmi-sound], device 0: ff8a0000.i2s-i2s-hifi i2s-hifi-0 [ff8a0000.i2s-i2s-hifi i2s-hifi-0]
  Subdevices: 0/1
  Subdevice #0: subdevice #0
```

then compared dts from image and found some related pieces:

```
rt5651-sound {
  status = "okay";
  compatible = "simple-audio-card";
  simple-audio-card,name = "realtek,rt5651-codec";
  simple-audio-card,format = "i2s";
  simple-audio-card,mclk-fs = <0x100>;
  simple-audio-card,widgets = "Microphone\0Mic Jack\0Headphone\0Headphone Jack";
  simple-audio-card,routing = "Mic Jack\0micbias1\0IN1P\0Mic Jack\0Headphone Jack\0HPOL\0Headphone Jack\0HPOR";

  simple-audio-card,cpu {
    sound-dai = <0xc4>;
  };

  simple-audio-card,codec {
    sound-dai = <0xc5>;
  };
};
...
rt5651@1a {
  #sound-dai-cells = <0x00>;
  compatible = "realtek,rt5651";
  reg = <0x1a>;
  clocks = <0x08 0x59>;
  clock-names = "mclk";
  pinctrl-names = "default";
  pinctrl-0 = <0x3c>;
  status = "okay";
  phandle = <0xc5>;
};
...
```

I'm not really into kernel development and think I could miss something.. right now `rt5651` is included only in `rk3399-sapphire-excavator` of mainline kernel

Could you push your `dts` patch for nanopi-m4?

avafinger, 2020-11-12:
Which kernel version did you build?

razum2um, 2020-11-13:
5.9.8

avafinger, 2020-11-14:
I don't usually maintain a patch set, here is the DTS patch to have wifi, BT, spdif, and rt5651, strip off the rest you don't want. It's been a long time since I worked on it.
[rt5651_spdif_wifi_bt_dts.patch.tar.gz](https://github.com/avafinger/nanopi-m4-ubuntu-base-minimal/files/5541555/rt5651_spdif_wifi_bt_dts.patch.tar.gz)

avafinger, 2020-11-14:
Just restore the following line to original one:

```
-#include "rk3399-nanopi4.dtsi"
+#include "rk3399-nanopi-m4.dtsi"

```