avafinger/nanopi-m4-ubuntu-base-minimal #29: is it possible to have a version of release 1.41 taht boots on emmc ?

hervald, 2020-11-13:
Hello,
first of all thanks for your great work. I used to work with a former release but now I would like to test with the most recent kernel. Unfortunately, the image you provide does not work as is if put on an emmc card. I tried to modify boot.src but it changes nothing : the m4is freezing, without launching the kernel.
So, do you know what I have to tune so that I can boot on emmc, or, can you provide an image that could be used directly on emmc ?
Thanks

hervald, 2020-11-13:
Hum, sorry for the previous message. I'm so an idiot. I recompiled boot.scr but I simply copied the original one the emmc, instead of the recompiled one. Sigh !
Indeed with the modified version, it boots correctly, even if I have a black screen surely because my monitor is too old (4:3 format)

avafinger, 2020-11-15:
You could pass a kernel command with the preferred mode through DDC/EDID.

video=HDMI-A-1:1280x720

or a specific edid:

drm.edid_firmware=edid/your_edid.bin
That should be in ls -la /usr/lib/firmware/edid/*.bin

hervald, 2020-11-15:
In fact, I seem to have the same problem as the one expose in a previous issue : the boot is stucked after "startgin kernel". I tried on different screen and it's always the same. So, I'm not sure taht it comes from a pb of display resolution.