avafinger/nanopi-m4-ubuntu-base-minimal #31: "green" images from cameras in 5.10/5.11

ayaromenok, 2021-03-29:
Not a issue - just for information ;)

By my understanding, color correction of this "green" images implemented in RkISP shared library (https://github.com/rockchip-linux/camera_engine_rkisp), based on information about sensor from IQ files (also in repo, in _iqfiles_ directory) - and therefore not exist at kernel V4L2 driver(and mjpg_streamer). This repo also contain *rkisp* gStreamer driver (which used in FriendlyElec example scripts) and C API (with example _apps/rkisp_demo_) - so MJPEG streamer can be updated to use RkISPlib/IQ files too.

avafinger, 2021-03-29:
There is no 3A fully implemented in the main 5.x kernel. Libcamera has some. The camera_engine_rkisp works only in 4.x unless you have ported it to 5.x, have you?

ayaromenok, 2021-04-13:
No, I have no reason to return to camera_engine_rkisp repo.

closed 2021-04-13 by ayaromenok