avafinger/nanopi-m4-ubuntu-base-minimal #33: Camera Parameters for ov4689

Mango-kid, 2021-06-08:
Hey, I have had good success working with the ov4689 but I am bashing my head against the wall right now trying to figure out the best path for controlling the camera parameters. 

It seems that in the driver there are only controls for some of the basics (ie, exposure and analog gain). Currently it looks like the best route is going to be expanding the driver to extend the functionality. For example with v4l2-ctl setting the exposure doesn't work correctly because auto exposure is always enabled. 

Short of modifying the driver is there anything you found for controlling this cameras parameters? Any help or direction would be great.

avafinger, 2021-06-08:
What kernel are you with?

Mango-kid, 2021-06-08:
The freindlyelec kernel-rockchip 4.4.179